iNavFlight/inav #4192: Disable support for DSHOT600 and DSHOT1200 for reliability reasons

digitalentity, 2019-01-11:
Per issue title.
Summoning @DzikuVx @fiam @stronnag @shellixyz for discussing this

DzikuVx, 2019-01-11:
I would not go that far, but I would add a HUGE red notification in Configurator saying "You are doing this on your own responsibility and probably should not do it if you are not ready to accept the potential consequences"

stronnag, 2019-01-11:
Agreed, and in the release notes (I added the target list already). We are in for a lot of these I think.

teckel12, 2019-01-11:
If we disable it, users will keep demanding it to be enabled.  Better to keep it with a warning.  Then, when issues come in (which we know they will) we'll just redirect them to the warning.

With that said, it seems that most of the problems with DSHOT are on the F7.  @DzikuVx were you testing DSHOT on the F7?

digitalentity, 2019-01-12:
Agreed. Let's make a warning in Configurator then. Moving the issue there.