iNavFlight/inav #4194: Matek F722-SE CPU% seems high

Nihilistic, 2019-01-12:
## Current Behavior

CPU% significantly higher on my Matek F722-SE equipped 5" inch, than on my identically configured Matek F405-CTR equipped quad.
## Steps to Reproduce

1. Flashed iNAV 2.1 RC1 to both Matek F722-SE and Matek F405-CTR
2. Applied quadx multirotor mixer
3. Applied 5" GPS preset
4. Selected MultiShot (with default ESC refresh)
5. Bumped LPF to 256
6. Bumped loop time to 4khz
7. Completed configuration without diffs or CLI pastes...
8. CPU% is 18% on the bench for F722-SE vs. about 9% for F405-SE.
9. Attempts to bump loop to 8 khz results in 1000% CPU on the F722-SE, but only about 40% CPU on F405-CTR
## Expected behavior

I would have expected the F722-SE to have lower CPU% on account of the faster processor...
## Suggested solution(s)

Any idea what's happening here?
## Additional context

---

- FC Board name and vendor:
- INAV version string:

digitalentity, 2019-01-12:
Output of `status` and `tasks` commands might shed some light on this

Nihilistic, 2019-01-12:
# status
System Uptime: 32 seconds
Current Time: 2019-01-12T03:10:57.399-06:00
Voltage: 0.0V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x1b, 15645696kB, 08/2016, v1.0, '00000'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20010000, Heap available: 1752
I2C Errors: 20, config size: 4315, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 11, cycle time: 251, PID rate: 3984, RX rate: 49, System rate: 9
Arming disabled flags: NAV RX CLI

Nihilistic, 2019-01-12:
# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9    3327       3    3.4%    0.5%     14370
 1 -     GYRO/PID    3891     201     128   78.7%   50.3%     61801
 2 -           RX      49      49      43    0.7%    0.7%       263
 3 -       SERIAL      98     513       3    5.5%    0.5%        43
 4 -      BATTERY      49      19      14    0.5%    0.5%        86
 5 -  TEMPERATURE       0       5       2    0.5%    0.5%         0
 6 -       BEEPER      99      11       4    0.6%    0.5%        52
 7 -          GPS      49     238      22    1.6%    0.6%       135
 8 -      COMPASS       9     216     213    0.6%    0.6%       261
 9 -         BARO      43     227     154    1.4%    1.1%       816
13 -    TELEMETRY     491     164       5    8.5%    0.7%       316
16 -          OSD     243    2026      54   49.7%    1.8%      1588
17 -          CMS      49       5       2    0.5%    0.5%        10
20 -      VTXCTRL      49       4       2    0.5%    0.5%         9
Task check function             5       2                        11
Total (excluding SERIAL)                   147.2%   58.8%

Nihilistic, 2019-01-12:
Does this tell us anything?

Nihilistic, 2019-01-12:
@teckl12, I'm at about 10% CPU on the F405-CTR with 5" GPS preset applied, multishot with defaults, and 4 khz loop time. And about 13% CPU on the F722-SE (the 18% I quoted earlier was when I had applied the 5" Performance preset...so more filtering and stuff on that one).

So, are you using a preset? What loop time?

teckel12, 2019-01-13:
@Nihilistic I'm using a GPS, barometer, stabilization etc., so of course I'm running 4k loop times.

Nihilistic, 2019-01-13:
@teckel12,

In my experience, the F7 is a step backward from the F4. It's actually slower. If I select 8 khz loop, the CPU% goes to 1000%. If I do the same with the F4, it goes to 40%. I don't actually want to fly with 8 khz. I guess I was assuming the F7 was a generation beyond the F4, and was therefore a more future proof option over upgrading to an F4. Seems like the only advantage gained over the F4 is that inversion is available on all the UARTs. I wasn't really having an issue with that. I'm using a Crossfire Nano RX on my F4 bird, so it only needs one UART. My F7 bird has the FrySky X4RSB, until I get Crossfire for it. Thinking of getting the new diversity Nano RX for it...but haven't seen any reviews or heard a peep about it on RCgroups yet. Don't feel like getting more early adopter arrows to the knee ;)

Nihilistic, 2019-01-13:
@digitalentity @teckel12 

The output of my "tasks" command says at the bottom "total maxload 147.2%". That's with loop time at 4khz. Does that mean that in the worst case scenario, it's possible that I could max out the CPU in flight and cause catastrophic failure, despite the fact that the CPU load is a mild 10% on the bench?

Nihilistic, 2019-01-13:
Here's the output of "tasks" if I drop back to 2 khz loop time.

Task list                     rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -            SYSTEM          9    3328       3    3.4%    0.5%      2665
 1 -         GYRO/PID    1960     194     128   38.5%   25.5%      3575
 2 -                     RX        49      49      42    0.7%    0.7%        31
 3 -               SERIAL       99     325       3    3.7%    0.5%         8
 4 -            BATTERY       49      18      14    0.5%    0.5%        10
 5 -  TEMPERATURE         0       3       1    0.5%    0.5%         0
 6 -              BEEPER       99      10       4    0.5%    0.5%         6
 7 -                   GPS       49     221      23    1.5%    0.6%        16
 8 -          COMPASS        9     214     211    0.6%    0.6%        30
 9 -                 BARO      43     226     152    1.4%    1.1%        95
13 -       TELEMETRY    493     163       4    8.5%    0.6%        48
16 -                  OSD    249    2018      50   50.7%    1.7%       160
17 -                 CMS       49       4       2    0.5%    0.5%         1
20 -          VTXCTRL       49       4       2    0.5%    0.5%         1
Task check function             5       2                         1
Total (excluding SERIAL)                   107.8%   33.8%

So, still over 100% even on 2k...my F405-CTR is at 90% maxload on 2k.

Either the target for this board is very poorly optimized, or this processor sucks ass.

DzikuVx, 2019-01-13:
OK, I think we can stop this discussion right now. I'm labeling this as bug that F7 can not reach 8kHz in INAV (this applies to other F7 boards as well) and until anything **new and value adding** appears, I see not reason to continue over here.
@Nihilistic Thank you for reporting

Nihilistic, 2019-01-13:
I think the new title is a bit misleading. This board seems to lag behind the F4 period.

And can you answer my questions about the output of the status and tasks commands?

DzikuVx, 2019-01-13:
@Nihilistic you are under wrong impression that this is this board specific and the new title actually describes _real_ reason in a wide scope

Nihilistic, 2019-01-13:
@DzikuVx 

So, this is a problem with all F7 boards? Or just the F722 chip ( the smaller chip). 

Does the output of the tasks command mean what I think it means? DE asked me to post it, and he never told me what it means. Am I not safe even running 2khz on this board?

leeph, 2019-01-18:
BetaFlight can run an 8khz loop on Matek F722-STD much higher looptimes on Matek F722-SE with the ICM20602 gyro, so it's not a fundamental issue with the F7 boards. I can confirm I see the +1000% CPU usage in iNav trying to run an 8khz loop time.

So I think Pawel is correct in categorising this as an iNav bug, actually.

Nihilistic, 2019-01-18:
CPU% is higher on my F722-SE on any loop time setting, compared to my F405-CTR with identical settings. Which suggests that the F722-SE has a weaker processor, or the target needs optimization. I'm hoping it's the latter, because there is much to like about the F722-SE otherwise. 

I can run my F722-SE at 4khz with 13% CPU and 5" GPS preset defaults (except for the lpf bumped to 256 and loop at 4 khz). But, DE said in another thread they're considering adding a warning that loop times of 4-8 khz is unsafe for navigation modes, after seeing my output of status and tasks commands on my F722-SE at 4 khz.

teckel12, 2019-01-18:
@Nihilistic I believe it's already been discussed ad nauseam that this is an issue with INAV having to do with F7 processor. It's not the F7 processor.

teckel12, 2019-01-18:
@Nihilistic Because it's been confirmed by the developers that enhancements to the F7 code could be made.  No additional discussion is required as far as I'm concerned.

DzikuVx, 2019-02-27:
with #4263 merged into development, problem will be fixed in next release of INAV (next after 2.1). This does not mean it will be possible to run 8kHz with GPS, MAG and BARO enabled. But F722 runs 8kHz for full acro purposes stable now

closed 2019-02-27 by DzikuVx