iNavFlight/inav #4195: Matek F722-SE iNAV 2.1 RC1 getting I2C error: 20 and -20 feet baro altitude in OSD

Nihilistic, 2019-01-12:
## Current Behavior

What does error code 20 on I2C mean? While hovering around the back yard yesterday, my altitude on my OSD was showing about -20 feet. If I climbed a bit, it would increase, but stay about 20 feet lower than it should. Baro calibration off or something?
## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

teckel12, 2019-01-12:
@Nihilistic Could be any number of things, but it just means the barometer sensor believed it went down 20 feet, which is not much. Maybe you didn't install foam on your barometer? Or, maybe a storm was coming in? Could be anything, but unrelated to INAV I'm sure.

Nihilistic, 2019-01-12:
@teckel12,

I've never bothered to put foam on any of my baros. Not had this problem with either of my Ominibus F3's, or my F405-CTR. I get that I2C error 20 when I plug into the board on the bench with the battery not plugged in. Thought there might be a table where one could look up what these codes mean.

OloTka, 2019-01-15:
I'm receiving the same I2C errors of 20 with MATEK F722 STD, does not matter if with battery connected or not. Anyway it looks like baro is working OK, so should that be something to worry about, mean those I2C errors? Btw, is that count of the errors or error code?

Nihilistic, 2019-01-15:
@OloTka Does your displayed (baro) altitude also seem off a bit? When watching the sensor tab, I've noticed that the altitude shifts significantly when my BN-880 gets a 3D fix and the red LED starts flashing.

With BetaFlight 3.5.4 on the board, the altitude output on the sensor tab would be normal, and then jump up 100 meters or something when my GPS got a 3D lock. I think they're mixing GPS altitude with baro altitude for the GPS rescue feature they're developing. I get i2c error 2 on BetaFlight.

OloTka, 2019-01-16:
@Nihilistic I don't the gps attached yet. only bt module on uart1. there seems to be no issues with baro with that minimal config I have. I'll try BF to see if those I2C errors appear.

OloTka, 2019-01-17:
Hello, uploaded BF into the board - got I2C errors = 6, then turned off baro and after restart got 4 errors. It looks for me like something is impacting initialization of I2C therefore those errors are on start of board only and they to not increase more. Anyone any thoughts?

digitalentity, 2019-01-19:
@Nihilistic calibration is a tad off in 2.0-2.1. Fixing that in https://github.com/iNavFlight/inav/pull/4202
The issue with baro altitude being offset by a certain value should be only a cosmetic issue. Also, it would be good to have a blackbox log with the issue - just to make sure the code is behaving as expected.

Nihilistic, 2019-01-19:
I posted a black box log with my first attempt at DSHOT 600 issue that apparently got closed or deleted. Go to the end and scroll back, because the file is like 10 minutes long...and the flight is the last 30 seconds or so. Had to go back inside and make a change, and come back out, which is why there's like 9 1/2 minutes of dead time at the start of the file.

Or I can post it again, if needed.

delamitry, 2019-01-31:
I'm also experiencing i2c bus errors with that target. The same '20' i2c error on startup and baro seems to work but if I add anything else to the bus (specifically an hmc5883l mag), MSP has trouble reconnecting, escs start toning out every few seconds, i2c errors continue to rise.

DzikuVx, 2019-02-27:
it looks like hardware i2c problem, unable to reproduce over here with f722se. Closing. Thank you very much

closed 2019-02-27 by DzikuVx

Nihilistic, 2019-02-28:
@DzikuVx With iNAV 2.1 full, the altitude displayed is correct. But, oddly, now I get i2c error 45. Everything works, so I just ignore it. I only mention it now in case it's helpful. You get no i2c error on yours at all?

You think there's a hardware error with the 722-SE or the BN-880? Could it be from excessively long wire leads from the BN-880 to the i2c pads for the compass? I left some slack so I could remove the top plate and set it aside without disconnecting the plug.

OloTka, 2019-03-01:
Hello, I've noticed the same i2c errors of 45 in my matek 722 STD, this is the change vs i2c errors of 20 vs previous FW. My GPS is not BN-880 so that I would exclude from the reason.

OloTka, 2019-03-01:
Hello, just noticed in other issue reported the same i2c errors 45 - "Bug in FW or wrong flight controller? #4427". Could be the same thing.

teckel12, 2019-03-01:
I have the Matek F722SE and BN-880 and haven't noticed any i2c errors when the battery is connected.  With just USB, there is sometimes not enough power to start the GPS and I get a navigation warning.  Maybe I get one i2c error on startup from time to time, but nothing out of the ordinary.  I'll check again to see if I get any errors with the latest INAV 2.1.0