iNavFlight/inav #4198: Enabling RX_MSP feature

tykimseoul, 2019-01-13:
Hi, I am trying to enable the MSP feature in the configurator.
My goal is to use the MSP protocol to make the flight controller and a Raspberry Pi communicate.

#### Flight Controller Model : Matek F405
#### Configurator Version : 2.0.0

I found some CLI commands for enabling MSP:
```
feature RX_MSP
set receiver_type = MSP
save
```
So I tried it:
[image]
then I saved and rebooted.

But when I reconnect and check features again, it does not list RX_MSP.
I think this might be why SET_RAW_RC command does not work.

Any ideas will be appreciated.

stronnag, 2019-01-13:
https://github.com/stronnag/msp_set_rx

tykimseoul, 2019-01-13:
@stronnag Thanks for your suggestion.
However, I have already checked out the link and did not succeed.

To elaborate on my issue, I have implemented the msp protocol and sent out the SET_RAW_RC command successfully.
I have even got a response with the same command code (code 200), which I believe is an expected result.
The only problem is that the motor does not spin.
So I was wondering if the feature RX_MSP of the configurator was the reason.

Shouldn't the RX_MSP feature stay even after I reboot?

Thanks

stronnag, 2019-01-13:
`FEATURE RX_*` was removed in iNav 1.8 (as stated on the previous link) and in the 1.8 release notes.

tykimseoul, 2019-01-13:
But the feature RX_MSP command still works.
It does not give an error and even gets listed in feature.
Do you mean that the command will have no effect?

stronnag, 2019-01-13:
Read the release note; draw your own conclusion ....

closed 2019-02-02 by shellixyz

tobetz, 2020-01-11:
@tykimseoul and @shlapion: Did you solve this problem? I would also like to control a copter with the RPi, and I am pretty much stuck where @tykimseoul was left. The feature RX_MSP is turning itself off. I have recompiled the firmware to enable MSP over UART. All no luck. I can send commands and get the 200 back, but I cannot arm and control the copter. 
The commands are sent with about 10Hz I think, but I have not perfectly measured. Any ideas how to convince the FC to answer. 

PS: I have a CC3D, and tried iNav and betaflight

stronnag, 2020-01-12:
> The feature RX_MSP is turning itself off. I have recompiled the firmware to enable MSP over UART. All no luck. I can send commands and get the 200 back, but I cannot arm and control the copter.
> The [...]

Notwithstanding that CC3D is long not supported, if you patch in `RX_MSP` it works just fine ... as long as you rigorously apply the required arming and update conditions (https://github.com/stronnag/msp_set_rx#caveats)

Here we go ....

```
$ ./msp_set_rx -A 5
2020/01/12 11:00:01 Using device /dev/ttyACM0
INAV v1.7.3 CC3D (772e015f) API 2.0
...
Rx: [1500 1500 1500 1000 1001 1442 1605 1669] (00300) unarmed Quiescent
Tx: [1500 1500 1500 1000 2000 1442 1605 1669]
Rx: [1500 1500 1500 1000 2000 1442 1605 1669] (00301) unarmed Arming
Tx: [1500 1500 1500 1000 2000 1442 1605 1669]
Rx: [1500 1500 1500 1000 2000 1442 1605 1669] (00302) armed Arming
...
Tx: [1500 1500 1500 1000 2000 1442 1605 1669]
Rx: [1500 1500 1500 1000 2000 1442 1605 1669] (00311) armed Min throttle
...
Rx: [1500 1500 1500 1000 2000 1442 1605 1669] (01500) armed Min throttle
Tx: [1500 1500 1500 1000 1000 1442 1605 1669]
Rx: [1500 1500 1500 1000 1000 1442 1605 1669] (01501) armed Dis-arming
Tx: [1500 1500 1500 1000 1000 1442 1605 1669]
Rx: [1500 1500 1500 1000 1000 1442 1605 1669] (01502) armed Dis-arming
Tx: [1500 1500 1500 1000 1000 1442 1605 1669]
Rx: [1500 1500 1500 1000 1000 1442 1605 1669] (01503) unarmed Dis-arming
...
Tx: [1500 1500 1500 1000 1001 1442 1605 1669]
Rx: [1500 1500 1500 1000 1001 1442 1605 1669] (01511) unarmed Quiescent
```