iNavFlight/inav #4200: Use rangefinder for ALTHOLD on low altitudes

SaschaIoT, 2019-01-13:
## Current Behavior
Only baro is used to hold altitude with ALTHOLD flight mode. Only in SURFACE flight mode the rangefinder is used to hold altitude.

## Desired Behavior
Use rangefinder in ALTHOLD to hold altitude to the max range of the rangefinder. Above the max range of the rangefinder use baro instead.

SaschaIoT, 2019-01-13:
@teckel12 I write only the idea (don't understand it as detailed specification), the developer has to test and decide how it works best.

teckel12, 2019-01-13:
@SaschaIoT I was just chiming in on how it would be implemented.

SaschaIoT, 2019-01-13:
@teckel12 

_I don't think you could use the max range of the sensor. First, the usable range is typically much less than the spec'ed max range._

Yes I also think max range can't be used, must be lower. Tested individually for each rangefinder.

_Second, who's to say your alt hold is at the same altitude as launch?_

I think with rangefinder it is only possible to have always the same distance to ground. So if you start the copter at low ground and fly to higher ground the copter goes up (but same distance to ground). But I think this is not a really problem. I think there sould be a CLI variable to define if rangefinder is used on low altitudes for ALTHOLD.

_Third, the baro could easily be off by a couple meters, which could be the entire real range of the sensor._

_It would seem that when alt hold is initiated, so would the rangefinder. If the rangefinder is receiving distance readings, then it would use the rangefinder for alt hold, regardless of the baros altitude reading._

Yes sounds good. On low altitudes and if rangefinder give good readings, rangefinder is used for ALTHOLD. In high altitudes (above rangefinder range) or in case of bad readings from rangefinder, baro is used instead.

closed 2022-12-04 by DzikuVx