iNavFlight/inav #4203: 2.1.0 RC1 Matek F411 mini hangs up upon the first reboot

Alejandro-pucci, 2019-01-13:
## Current Behavior
Flashing 2.1.0 RC1 to Matek F411 mini causes a hang up upon the first reboot. Meaning you can get to the configurator once. I tried the 2.1.0  RC1 regular version and the 2 soft serial version. 

Other behaviour:
In the 2 soft serial version the ports page lists the ports as follows:
USB VCP
UART1
UART2
SOFTSERIAL2
UART1

Also the OSD via the VTX was dimmer than usual and blinking.

- FC Board name and vendor: Matek F411 mini (MK41, version 0)
- INAV version string: 2.1.0  RC1 regular version and 2 soft serial version
- CONFIGURATOR 2.1.0

teckel12, 2019-01-13:
@Alejandro-pucci Related to https://github.com/iNavFlight/inav/issues/4183 ?

Good advice about doing a full erase when flashing. Also, uploading the fonts.

Alejandro-pucci, 2019-01-14:
Hi, thanks for the reply,

The description of the blinking osd matches #4183, but the bigger problem is that I can't reboot the board and connect to the configurator.

Full chip erase was enabled. Uploading the fonts causes a reboot after which I can't get to the configurator. Everything worked fine in 2.0.1 2 soft serials from this thread:https://github.com/iNavFlight/inav/issues/3919

I have a different probably offtopic (unless there is something also in the aforementioned .hex) question: Can you not use a soft serial port for an ibus receiver?

Xedos9er, 2019-01-18:
I noticed also some weird behavior. When flashing 2.1.0 on my F411 Wing i see in boot screen something like "missing fonds" right on top of the info text "INAV 2.1.0... bla bla bla". When I upload new fonds in the OSD menu of the configurator, I see on boot a multicopter logo. After that I can´t get into the OSD configuration menu in the configurator, it becomes very sluggish and doesn´t load the setting, screen is just blank. Sometimes it load the preview without the options, after clicking on the fonds button all options start to appear. When you don´t upload fonds after a flash, the OSD menu works without issues in the configurator till you upload first time fonds?!
The same behavior can be seen on my F405Wing.

teckel12, 2019-01-18:
@Xedos9er What's the CPU load?

Xedos9er, 2019-01-18:
no more than 5-10%, already checked it. issue is reproducible.
it has something to do with the fonds. If you flash and don´t load any fonds, no issues. But there´s no quadrocopter bootlogo, only the text

noagendafpv, 2019-01-23:
I loaded the 2 soft serial FW to my F411-W last night and tried to setup the ports for SmartAudio and SmartPort. The ports tab shows UART1, UART2, SoftSerial2 and UART1 again at the bottom. I assumed the second UART1 is the Soft Serial 1, and set SS2 to S.Port and Uart1(Bottom of list) to SmartAudio. Saved and rebooted and then the FC would not communicate with the iNav configurator. I tried restarting the config, tried a different USB cable, tried restarting my laptop and none of that worked. I then reloaded the stable 2.0.1 FW and the FC was back to functioning properly.

shellixyz, 2019-03-01:
I can't reproduce the issue with 2.1.0 final and configurator 2.1.4. The issue with softserial have been fixed by #4315. Please test with 2.1.0 and configurator >= 2.1.4 and open a new issue if it persists.

closed 2019-03-01 by shellixyz