iNavFlight/inav #4207: Settings Validated Red X in 2.1 RC1 and Battery Voltage Displays No Decimals

Nihilistic, 2019-01-15:
## Current Behavior

After setting up the OSD on both my F405-CTR and F722-SE, and plugging in a flight battery for the first test flight, I got a navigation is unsafe voltage something or other error flashing on my OSD, and only the whole numbers of my battery voltage would display. Workaround is selecting 2 decimal places from the dialog box on the OSD tab, clicking save, and then selecting back to 1 decimal place, and clicking save, and then voltage displays correctly on OSD.

Possibly related to #3507
## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

giacomo892, 2019-01-15:
If settings are no validated you likely have not upgraded following the guide on the release notes.

Nihilistic, 2019-01-15:
I always do a full chip erase and do the full configuration manually. No CLI dump pastings at all.

After clicking on the dialog box and bumping decimals up to 2 and clicking save, and then selecting 1 decimal again and clicking save, the settings validated issue goes away. Only have to do this once, not every time I plug in the battery.

It's a minor thing, but thought I'd mention it.

DzikuVx, 2019-01-15:
[image]

[image]

Just tested on F405-CTR and it is working correctly with Full Chip Erase, settings are valid after flashing

closed 2019-01-15 by DzikuVx

Nihilistic, 2019-01-15:
I'm using the modified hex that fixes the Split 2S control issue that was just posted today. But, I've had this problem before using the official 2.1 RC1 hex. Maybe it's something very specific to how I'm setting up my OSD (which fields I've chosen and where I've placed them). Here's my most recent dump, but it's after I did the workaround, so I don't know if it will show anything.
[Fully Configured Dump.txt](https://github.com/iNavFlight/inav/files/2761573/Fully.Configured.Dump.txt)