iNavFlight/inav #4210: Velocity vector/angle of attack

Fluburtur, 2019-01-15:
As far as I know, there isn't any way to see the velocity vector or angle of attack in Inav yet. I think this feature would be useful for plane pilots as it would allow them to reduce the angle of attack while flying, increasing efficiency and maybe see a stall coming.

I am not very good at maths but for what I know it could be done with the IMU alone, substract gravity and see what forces are effected on the plane and this way, see it which direction it is traveling.

To explain it better, I have drawn what I would like on a screenshot I took of a flying session.

[image]

As you see, the crosshair is the direction the nose is pointed but the glider is gliding and descending slowly so the actual direction the glider is flying is shown by the velocity vector. This is widely implemented on pretty much any plane equipped with a HUD and given the OSD of my glider is more crowded than that of a fighter jet, it wouldn't be out of place there.

If this gets implemented it would also be possible to add an angle of attack limiter and display, as well as sideslip display.

As I said, this would benefit anyone looking for higher aerodynamic efficiency and glider pilots, perhaps it could work on multirotors as well but I dont fly much of those so I can't say for sure.

DzikuVx, 2019-01-15:
I do not think we actually can measure AoA only using IMU. Any acceleration caused by turn would show up on the ACC readings and this acceleration would also would have to be substracted. Unfortunately we do not know this acceleration.

Probably it can be "approximated" using GPS 3D heading substracted from MAG heading. Or just 3D speed comparing to ground speed. But then it would be only 2D and not really an AoA but only velocity vector.

Bigger airplanes have AoA measurement devices. We do not have them.

rb1205, 2019-01-15:
You can get a good approximation of the vertical component of the true speed from the differential of the baro reading. GPS heading gives you the horizontal components with good precision, and that should do it. It'd still need a mag to get the plane heading tho.

wojciechwas, 2019-01-16:
Would definitely be very useful. The flight path vector would be much appreciated. 

I already suggested this in the past:  
#3578
https://github.com/iNavFlight/inav/issues/3578 
It got pushed from 2.1 to future milestone at the moment. 

As some of the devs mentioned the biggest challenge/killer is the character based OSD, even if we can calculate it.  
Unless a bunch of characters were added to the osd to represent pixel level the flight path vector this feature would be difficult to implement. 
If only the OSD could be pixel based... 

It might be worth to tie up with @teckel12 and try out this feature with the glass cockpit through lua telemetry. He mentioned in the past he might be able to add this feature.

closed 2024-02-02 by DzikuVx