iNavFlight/inav #4214: Airbot NOX support

mooiweertje, 2019-01-16:
I've created a new target for my NOX compliant board. I'd like to share it so it can be part of the next release. Can I just push it ?

teckel12, 2019-01-16:
@mooiweertje Create a pull request to the development branch.  Additional information: https://github.com/iNavFlight/inav/blob/master/docs/development/Development.md

mooiweertje, 2019-01-16:
First time doing this but I think I succeeded.

mooiweertje, 2019-01-16:
Done: https://github.com/iNavFlight/inav/pull/4216

shellixyz, 2019-02-02:
PR to add the board staging, closing

closed 2019-02-02 by shellixyz

mooiweertje, 2022-10-07:
I created a version where GPS can be connected to PPM and LED via Soft Serial 2

[jnav_5.0.0.0_NOX.zip](https://github.com/iNavFlight/inav/files/9735545/jnav_5.0.0.0_NOX.zip)