iNavFlight/inav #4217: Switch to OSD page that alarm is occuring on

kardon18, 2019-01-17:
## Desired Behavior
Due to there being up to 3 OSD pages that can be selected. 
Very often, different information will be displayed on each layout page.
 
So if an alarm occurs on an OSD page that is not currently displayed. Then we have no idea that an alarm is occurring.

## Suggested Solution
When ever an alarm occurs. Switch to the OSD page that the alarm is displayed on.

This could have a time related switching period. 
e.g. It would switch to that OSD page that has the displayed item alarm for 5 second (long enough to get the pilots attention). Then switch back to the OSD page that it was originally on.

## Who does this impact? Who is this for?
Anyone who wants to be warned of critical information, under any condition. Low RSSI alarm, Low battery capacity etc.

wx4cb, 2019-01-17:
personally that could get distracting, but it's a good idea. maybe instead of switching it, then flashing a system message instead?

Obviously this would have to be configurable as to whether you want it or not... personally I don't have any alarms set, and I always have the important info on all the osds i use (voltage/mah/rssi/lat-long)

kardon18, 2019-01-17:
@wx4cb 
I agree. Configurablity would be a must.
The explanation, was only a possibility of the setup. I do agree that it could be distracting if a particular alarm was consistently being triggered.

AS you said, the system message could display the alarm. As it does in some cases.

Or possibly having the specific symbol that would display the alarm, appear (and flash) in a preset location on any OSD page that is present at the time. 
Then disappear from the OSD page your on; after the alarm stops. Rather than switch the whole page.

wx4cb, 2019-01-17:
@kardon18 yea having a symbol flash could be an idea. the osd is character based so you have to take that in consideration

closed 2019-04-01 by stale[bot]