iNavFlight/inav #4218: Taranis X9D Plus X8R XSR Sensor lost.

DonaldHillNZ, 2019-01-17:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

teckel12, 2019-01-17:
@DonaldHillNZ Does this happen on the bench or while flying? Due to heat, I get sensor lost messages in the summer or on the bench. While flying it's never an issue due to air cooling. I believe it's a common problem with FrSky receivers placed too closely to VTx.

DonaldHillNZ, 2019-01-17:
On the bench Tim but Ive removed everything so no VTX connected and a 
cooling fan playing over the equipment and its definitely not at all 
hot. Everything is not only cool its cold to touch.

<mention@noreply.github.com>
Sent: 17/01/2019 5:13:26 PM
(#4218)

Nihilistic, 2019-01-17:
What version of OpenTX are you running?

DonaldHillNZ, 2019-01-17:
Ver 2.2.2

Nihilistic, 2019-01-17:
2.2.3 is out. Might be worth a shot upgrading.

DonaldHillNZ, 2019-01-17:
great thank you ill give that a shot. Just cant find any other ideas

Sent: 18/01/2019 11:52:02 AM
(#4218)

DonaldHillNZ, 2019-01-18:
Problem solved by upgrading the firmware on the Taranis to 2.2.3. At least its run for nearly 25 minutes with NO sensor lost.

Nihilistic, 2019-01-18:
Yay!!

Dronitoni, 2019-01-23:
here's a workaround from J.B. to that Bug:

https://www.youtube.com/watch?v=d-WSy4odpVE

teckel12, 2019-01-23:
@Dronitoni That really doesn't fix the problem, just removes the alert, which is an important piece of information to some (typically means you're going out of range).

closed 2019-02-06 by DzikuVx