iNavFlight/inav #4222: Almost no stick control in ANGLE mode

dronus, 2019-01-18:
## Current Behavior
Stick control is very low in ANGLE mode using 2.1.0-RC1 using default racing setup. 
The sticks have to be pulled really hard to achieve some degree of inclination on pitch and roll despite reasonable settings. There in fact is some control, but it is extremely low. Yaw angular is somewhat low too, but almost usable.
Also tried with several synchronized and unsynchronized gyro / loop timings, as those are described to possibly impede ANGLE mode performance.

## Steps to Reproduce

1. Use a miniquad with "racing" preset and hardware as seen in diff attached (MPU6000, PPM receiver)
2. Activate ANGLE mode
3. Try to fly around (better indoors, won't handle winds)

## Expected behavior
Quad should respond to stick controls in a default sensitive fashion, especially full stick deflection for pitch or roll should match about 40 degree inclination, as the parameters 
`max_angle_inclination_rll` and  `max_angle_inclination_pit` are set to 400.

## Suggested solution(s)
I think there is a scaling bug in some calculations, or the PID is fighting against control input somehow.
But maybe it's just the racing preset's combination of PID settings that made ANGLE mode sluggish, as most racers will of cause fly without ANGLE mode so it's maybe not tested. 

## Additional context

[inav_racer_dump.txt](https://github.com/iNavFlight/inav/files/2771458/inav_racer_dump.txt)

---

- FC Board name and vendor:
EMAX MAGNUM F4
- INAV version string:
INAV/OMNIBUSF4 2.1.0 Jan 18 2019 / 01:23:42 (f8de2e6a3)
GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

giacomo892, 2019-01-18:
Is the radio well configured to output 1000-2000 on all channels?

dronus, 2019-01-18:
I get about 1100 - 1900 from radio, so not full range but almost. Can't be configured, as it is an older FM transmitter. However this should sport about 80% range (or 60% with expo considered) of controll, which should be enough to cope with small winds... 
Currently even a breeze need almost full range of control to cope with. 
I am playing now at the INAV Configuratior imposed limit of 90 degrees and level P at 80, which already give oscilations while grounded, so this is the all maxed out in favor of control.  In Betaflight I've got a very nervous control even with far more conservative settings. In INAV, I got almost no control on Pitch Yaw and Roll, only throttle is nervous as expected for miniquad.

digitalentity, 2019-01-18:
```
set max_angle_inclination_rll = 400
set max_angle_inclination_pit = 400
```
Limits are 40 deg. Combined with expo and 80% range this will give like 25-30 deg max angle.
Level gain is also at default 20 which is lazy and smooth. 
Steps to fix:

1. Use `rxrange` to map your radio ranges to [1000;2000] (alternatively, get a modern radio)
2. Set tilt limits to something more reasonable like 60-70 deg
3. Tweak Roll/Pitch/Yaw PIDs to make sure quad is responsive in acro
4. Tweak level gain to get snappy control in angle.

closed 2019-01-25 by DzikuVx