iNavFlight/inav #4223: Inav 2.1.0 conf + F405-wing - S9 servo output

jreis1, 2019-01-18:
Hey there, 

Inav Configurator 2.1.0, Inav 2.0.1,  matek F405-wing. 

How can I activate (use) output S9 in matek board?
Instruction in Matek website are quite clear and say to enable: "Forward aux channels to servo output", as follows:
[image]

However, this option is no longer available in the current GUI version. 

If I go to mixer Tab, and try to had a new servo, it is indeed added to the servo list: 
[image]

but as soon as I save&reboot, and return back to the servo mixer tab, the newly added servo is no longer in the list:

[image]

Any ideias?

jreis1, 2019-01-29:
> Looks like a bug

Hey, 
I've found solution for this.  I had to clear all the servo mixing, and add the servos again to the list (using add new mixer rule),  this time starting by Servo 0, up to Servo 6.

shellixyz, 2019-01-29:
Good news that you found a solution but still it is a bit strange

DzikuVx, 2019-02-27:
Firmware allows for max 8 servos indexed from 0 to 7. I see you had index 8 so it was rejecting it.

closed 2019-02-27 by DzikuVx