iNavFlight/inav #4225: smartport telemetry does not work on Wing FC-10 DOF using "FuriousFPV F-35" target in 2.X

dangerg, 2019-01-18:
## Current Behavior
Smartport does work on 1.91 but does not work on 2.01 or 2.1.

## Steps to Reproduce
Try smartport on 2.X and you will get no data.
I have tried:
  1.91: Works
  2.01: No Data
  2.1:  No Data

## Expected behavior

## Suggested solution(s)

## Additional context
Settings:
https://pastebin.com/3TYd0ymr

---

- FC Board name and vendor:
Wing FC-10 DOF
The receiver is an frsky XSR

- INAV version string:
 INAV/FF_F35_LIGHTNING 2.0.1 Oct 31 2018 / 18:53:37 (a04e68aaa)

DzikuVx, 2019-01-18:
1. This is not officially supported board
2. With INAV 2.0 the meaning of `telemetry_inverted` has changed. Try `ON` instead of `OFF`
3. How do you connect RX telemetry? This board has inverters? RX hack?

dangerg, 2019-01-19:
I tried again with telemetry_inverted = ON. Does not work.... 
This board is a clone of the F35_LIGHTNING.
It has a dedicated inverted SportPort output (TX6):

https://www.banggood.com/Wing-FC-10-DOF-Flight-Controller-INAV-OSD-Accelerometer-Barometer-Gyro-Compass-For-RC-Airplane-Drone-p-1318626.html?cur_warehouse=CN

same as the F35:
https://furiousfpv.com/product_info.php?products_id=657

Once again it does work with 1.91

PS: INAV rocks. You guys do an amazing job.

dangerg, 2019-01-20:
I figured out that:
` set smartport_uart_unidir = ON
`is also required.
Works fine now on 2.X

closed 2019-01-20 by dangerg

fiam, 2019-01-20:
@dangerg `smartport_uart_unidir = ON` is set by the default configuration for that board.

supersauri, 2019-01-24:
How did you manage it to work?

dangerg, 2019-01-24:
Turn on both
smartport_uart_unidir = ON
telemetry_inverted = ON

dickfranklinhsv, 2022-01-21:
Upon connecting to a 3S LIPO, the '3.3v OK' LED started to smoke, and the 3.3v pin was 0.0v.  So, not usable for RXs needing 3.3v.