iNavFlight/inav #4226: Wind gust correction / general external forces correction

dronus, 2019-01-18:
## Current Behavior
If hoovering with manual control (with or without ANGLE mode), wind gusts will drag the quad around, demanding pilot counter action. 

## Desired Behavior
If a short wind gust accelerates the craft in any direction, it should counteract by itself, canceling out the wind motion by adjusting roll / pitch / throttle. 

## Suggested Solution
Add a new mode that can be toggled while using manual control modes like ACRO or ANGLE mode (but maybe even useful in autonomous modes).
When activated, three PIDs go in place that controlls yaw, pitch and throttle by the corresponding accelerometer values, trying to to keep X and Y at zero and Z at one or an throttle dependend value (here it gets tricky). This way only the motor forces are left to move the quad, and every external force, like a sudden wind gust, will be canceled out. 
Permanent exposure to winds will of course still build up some long term movement, which we can not cancel this way, but this is left for the higher level GPS advanced modes.

## Who does this impact? Who is this for?
Anyone who does not go acrobatic but just want a smooth flying craft, keeping cruise or position with minimal stick involvement.

## Additional context
Actually the UAVP project had implemented this about ten years ago, there was a simple mixing of the accelerometer signals into the main pitch and roll PIDs with configurable weights, that even was permanently engaged, if configured to non-zero. It helped a lot to quite out the quadrocopter movements in outdoor conditions, and even seem to reduce the proximity effect in indoor environments.

wx4cb, 2019-01-18:
isn't this position hold ?

dronus, 2019-01-18:
Hey, this feature is not equivalent to POSITION HOLD. POSITION HOLD needs GPS and MAG, as you said, and POSITION HOLD controls by absolute position and velocity derived from absolute position (GPS). This gives a kind of slower, not so tight control of absolute position and velocity.

My suggestion described a more low-level approach just eliminating transient (eg. gust) movement, works relative to the crafts coordinate system, gives an almost instant reaction to calm down movement. 
It can be combined with almost any other flight mode, while beeing most useful for manual flight. But it may still calm down POSITION HOLD too if engaged, as it can react to external forces before substancial velocity, and thus a change in absolute position builds up that will be detected by GPS later.

dronus, 2019-01-18:
Also as I stated, this was already implemented over 10 years ago on 8-bit controllers, calming down movement without any GPS available. It also benefits indoor flight.

wx4cb, 2019-01-18:
wouldn't that be part of pid tuning? if you're in manual or angle mode you're still going to be affected by gusts. without a gps there is no coordinate system for the aircraft to know where it is in 3d space.

pid tuning is what you would need to do to calm down any other movements. or am i missing something ?

giacomo892, 2019-01-18:
How do you know it was wind or wanted motion without GPS?

dronus, 2019-01-18:
All wanted motion is done by thrust, and for multicopters thrust is always "straight up", and gives an accelerometer signal of almost (0, 0, thrust)g's.  Any external force making applying a not-thrust-direction acceleration is thus detected directly by the accelerometer. Any external force in thrust direction is detected by the difference between wanted thrust and the accelerometer Z value, however this is more involved as we have to know our wanted acceleration, which is usually given by a certain throttle signal. 

I think the attitude PID, as far as the configuration speaks it, only relies on attidute information, the gyros.
The control therefore counteracts to all forces changing the roll, pitch or yaw of the craft, which is fine to control the angle by RC.

It does not, however, react to external linear motions of the craft, as those resulting from gusts, propeller wash or proximity effects if approaching walls or obstacles. 

But linear external forces will give a clear trace in the accelerometer. In an idealized view, if there are no external forces, the accelerometer will only see thrust after leaving the ground. If it is well installed, this gives for example a vector like (0,0,9.81)g  while hoovering. Any external force, like a sudden wind gust, that accelerates the craft into a non-thrust direction, will show up as deviation to (0,0,.9.81)g immediately. So we can cancel out this external force by actively steering the accelerometer signal to (0,0,9.81)g. This can be done by three additional PID controllers, who try to keep the ACC zeroed by generating signals similar the sticks would do, or even more simple by directly mixing ACC signal into the main roll, pittch PID (works only 2D this way, doesnt control thrust).

giacomo892, 2019-01-18:
ACC is very noisy and drifts over time. An absolute device, like GPS, is required.

dronus, 2019-01-18:
For force-based counteraction, there is no integration needed. Force leads to acceleration, acceleration forces the counteraction. After the external force is gone, the controller doesn't interfere anymore. 

We will be using acceleration, eg. change of velocity, eg. change of change of position for the control. Not velocity or even position!

dronus, 2019-01-18:
Imagine you try to push the craft by the side. Your finger's force will cause some acceleration which then will be fed back to bank the craft against your push, opposing your finger. Any speed the craft has gained before its reaction will still be there (as ACC is zero again after the correction), so ANGLE mode will try to level the craft out again. This may lead to another acceleration, if the finger is still there, so basically we get a balance between leveling and external force counteraction. Even without integration, this will still remove some of the effects of your finger. It won't counteract velocity gained due to imperfect counteraction, nor would it hold an absolute position in the long term. Don't try this with finger...

Even if we would do integration of acceleration (which gives us velocity) and counteract on this, thus steering the long term accelerometer values to zero, a drift of accelerometer baseline would only add some pitch, roll or yaw deviation which would just add up a little on legacy deviation to perfect level if flying in ANGLE mode, and thus can be trimmed out likewise.

digitalentity, 2019-01-18:
TL;DR Not possible with hobby-grade sensors.

You want quad to cancel external disturbances effectively holding position. This is pretty much a definition of POSHOLD mode. For clear skies - use GPS. For GPS-denied environments we'll have optic flow support soon enough - it's in progress. It's not possible to do position hold with only accelerometer. Our sensors are not accurate enough, our sensor alignment is usually a few degrees off, motors are shaky, frames resonate - there are a whole bunch of reasons why we need a better source of truth for holding position.

I'll consider this issue closed and won't spend any more time discussing this.

dronus, 2019-01-19:
Maybe I have time to implement the next days, however I am quite new to INAV to may take some time to get used to the code.

dronus, 2019-01-19:
> TL;DR Not possible with hobby-grade sensors.

-Was done over 10 years ago using first affordable MEMS on market (today ones are much better)
-had significant effect on hoovering smoothness 

> You want quad to cancel external disturbances effectively holding position. This is pretty much a definition of POSHOLD mode.

No, I just want to cancel out (most of) transient acceleration by external disturbances. Wont' hold position nor velocity, just make unwanted change in position and velocity much harder.

It is like the gyros, they are making a change in attitude by external forces much harder.