iNavFlight/inav #4228: rth in case of failure

tiriad, 2019-01-18:
## Current Behavior

1. The rth have failed again, the first time i have lost a quadcopter where the compass didn't go on a mast and I thought this was the error, this time i already had installed a gsm locator, in a previous battery i had used poshold and rth successfully, iNAV 2.1 omnibusf4v3 version 
failsafe video failure [(https://youtu.be/joE4OhFKaso)]
The compass has worked for me with CW270flip alignment in previous versions and the headings were correct and it was installed on a mast to avoid interference.
I had the following variable configured like this,  rth_abort_threshold = 5000, enough to detect that the heading was incorrect that's why it landed after failing

2. When the **emergency landing** is made and it touches the ground the motors keep on spinning consuming unnecessarily battery and exhausting it losing any sonorous warning or video signal quickly if anyone's around is near if you do not have an autonomous buzzer.

## Desired Behavior

1. When the compass fails the quadcopter should have an additional check that exceeded this limit will use another heading as the gps for rth, similar to this feature and at least get close to you until you recover the video signal. **GPS heading**: #https://github.com/iNavFlight/inav/issues/2688
**nav_rth_abort_threshold** : it should have a preset for quadcopter of **5000** cm or less to prevent it from going uncontrollably far away as the change of direction in a quadcopter is immediate with a turn of yaw.

2. In an emergency landing if after a certain period of time there is no further descent of height the **motors** should be **switched off** to prevent burning if hooked on branches or battery runs out.

_There should be an alternative method to prevent your quadcopter when in longrange fail its compass knows to return to you without losing it_

kardon18, 2019-01-21:
@tiriad In an emergency landing if after a certain period of time there is no further descent of height the motors should be switched off to prevent burning if hooked on branches or battery runs out.

Flying/not flying detection #2331

Improg, 2019-03-28:
Waiting for this feature too. I use compass without a mast, so with li-ion batteries RTH fails often, with 4s2p li-ion - always fails, lipo batteries works better, because they are not so magnetized.

ulph0, 2023-04-13:
hi, i do have the same feature request. i am also flying without a mast, because of the different flying styles i have i had to regularly remove it. everybody i know who flies longrange is using betaflight because of that. but i like inav and the possibility to have precise navigation with compass, but would also like to have the chance to fly longrange just with gps,. and with li-ion. as an option. maybe also li-ion batteries with shells not out of steel could be a solution but i don't know if there are any.

Improg, 2023-04-14:
Hi,
Li-ion is much worse than lipo batteries because they are usually magnetised from factory. They have steel parts inside. I don’t use mast and use li-ion in longrange flights, and iNav, li-ion batteries have to be demagnitized before use.
 
Best regards.

> hi, i do have the same feature request. i am also flying without a mast, because of the different flying styles i have i had to regularly remove it. everybody i know who flies longrange is using [...]

ulph0, 2023-04-26:
that sounds like good solution. would you mind to tell me how you demagnetize a li-ion pack?

Improg, 2023-04-27:
It can be any method available to you, generating variable magnetic field, gradually decreased. Search for demagnetizers for old kinescopes, for example.

> 
> 
>that sounds like good solution. would you mind to tell me how you demagnetize a li-ion pack?

closed 2024-02-02 by DzikuVx