iNavFlight/inav #4237: Rx input broken on quanton target

PTDreamer, 2019-01-20:
PWM and PPM Rx input has been broken on quanton since e4f31fb. The pins are being configured as regular inputs instead of AF.
I was going to open a pull request with the simple fix but I see those files were heavily refactored on the dev branch, haven't taken the time to see if the issue is still there.

Thanks

shellixyz, 2019-01-29:
Quanton target ? I don't see it or anything close

PTDreamer, 2019-01-29:
https://github.com/iNavFlight/inav/tree/master/src/main/target/COLIBRI

shellixyz, 2019-02-02:
I don't have one of these boards. Please try 2.1.0-RC2 and report whether the issue is still there or not.

shellixyz, 2019-03-01:
Have you been able to confirm the issue is still present in 2.1.0 ? If you know how to fix the issue please submit a pull request.

closed 2019-05-15 by stale[bot]