iNavFlight/inav #4238: Looking for fc that is not Omnibus V6

tehawk, 2019-01-21:
Hello,

Please let me know which FC would be most painless for the following:

4in1 ESC Racerstar Aniversary Edition 35A
BN-880
FS-IA6B

I had Omnibus F4 AIO V6, got it cause it was supposed to be supported, but it was not. Eventually I burned it after connecting battery and I want to avoid it. If you ask me this fc cannot work with VBAT, also it says that it uses target Fireworks V2 but that is not true either.

teckel12, 2019-01-21:
@tehawk I'm suggesting the Matek F722SE for new builds. More IO ports, fully supported by INAV, fast, rock-solid, future-thinking (ready for blheli_32 telemetry for example), etc. And since the F7 supports unverted UARTS, even tricky FrSky builds with different protocols (for example F.Port) are supported.

I've been very happy with it and probably will use it for future builds till something drastically better comes out.

teckel12, 2019-01-24:
@tehawk I don't use 4 in 1 ESCs for 30mm builds so I'm not sure. The F722SE is designed for 4 individual ESCs, which is my preferred method to avoid interference.

closed 2019-01-25 by DzikuVx