iNavFlight/inav #4239: INAV wont load after flash on RXSR-FC (OMNIBUS F4 Fireworks V2)

TIMO88D, 2019-01-21:
## Current Behavior
RXSR-FC (OMNIBUS F4 Fireworks V2) will not open in inav

## Steps to Reproduce
have a RXSR-FC (OMNIBUS F4 Fireworks V2)
1. enter boot mode on FC by holding button
2. Connect to windows 10 pc via usb (ensure correct driver is elected through Zadig (winusb for stem32))
3. use Inav configurator to flash FireworksV2 Firmware succefully
4.  reboot fc
5. attempt to connect through correct com Chanel (3 for me)
6. "Serial port successfully opened ID:1"
7. "no configuration received within 10 seconds, communication falied"
8. give up, re flash betaflight, open bug issue. 

## Expected behavior
Flash will work, but Inav configurator will not be able to open a serial connection

## Suggested solution(s)
Unknown

## Additional context

---

not possible

- FC Board name and vendor:
RXSR-FC (OMNIBUS F4 Fireworks V2) 

- INAV version string:
Inav configurator 2.1.0, FrieworksV2 firmware 2.0.1

a1axx, 2019-01-27:
Same behaviour with omniNXT f7

bchristal, 2019-02-02:
Same issue with Matek F22 and Omnibus F4. The Omnibus F4 connects fine with Betaflight. 

Windows 10
CP210x USB to UART Bridge Virtual COM Port Windows 10 Universal (v10.1.4) - Installed

[image]

Device Manager:
[image]

PHMT, 2019-04-14:
Same behavior (and error) as the OP.  Trying to use 2.1 on the Kakute F7 (non-AIO version) using INAV configurator 2.1.4.  Re-flashing Betaflight 4.0 works just fine.

closed 2019-06-28 by stale[bot]

Mougs, 2019-10-04:
Hello,
Nobody have a solution for this communication problem?