iNavFlight/inav #4244: 2.1 RC1 Smart audio on F35 Lightning with AKK not working

wx4cb, 2019-01-21:
Latest RC candidate with F35 FC Full options.

AKK x2 ultimate hooked up and working, hooked up the smart audio to UART 4 and set to TBS (also tried tramp) and all the other baud rate options aren't working.

Seems to be the same as the following
https://github.com/iNavFlight/inav/issues/3753
https://github.com/iNavFlight/inav/issues/3681
https://github.com/iNavFlight/inav/issues/3625

This was fixed in 2.0 if i remember correctly, but i've tried 3 akks and also the RDQ mach2/3 which are effectively the same (iirc) and they all just get scrambled displays on the 7 segment display on board.

any method to change it via the osd fails, just says the stats page where it's trying to send etc.

Any thoughts as to where to check, the VTX SA works on betaflight.

wx4cb, 2019-01-21:
> It's probably the bug fixed in #4231

Cool. Cant wait for rc2 hehe

shellixyz, 2019-02-02:
Please report if RC2 fixed the issue

dangerg, 2019-03-21:
Same issue here. I have tried both the AKK and the TX805. I will try with other boards.

dangerg, 2019-04-28:
FYI: Tried the same AKK and TX805 on a F4 flight controller with no problems. It appears to be a problem with the F35 and smart audio.

sniperxnl, 2019-09-03:
Hi Guys,

Any news on this?
we have a customer with a F35 and AKK vtx and don't get it working.

sniperxnl, 2019-11-12:
Any news here guys?

dangerg, 2019-11-12:
Try softserial.

bigbear2001, 2019-12-26:
For TX4, try changing this resistor from 1 KOhm to 100 Ohm. It helped me solve this problem on Wing FC 10 DOF
[image]

datamann, 2019-12-26:
I have the same issue, AKK Smart Audio on TX4 on FC Wing. @bigbear2001 Which resistor did you change?

dangerg, 2020-03-06:
> I have the same issue, AKK Smart Audio on TX4 on FC Wing. @bigbear2001 I will try the same to see if it helps.

> For TX4, try changing this resistor from 1 KOhm to 100 Ohm. It helped me solve this problem on Wing FC 10 DOF
> [image]

This is so small. Are there any other pins (like the header pins) that would work?

modwadd, 2020-04-06:
Hello, I head the same issue.
if the vtx is enabled the signal on tx4 loks good, but if  the tx804 V2.0 is powered up the signal is going down to low much to slow (i measured it by an osci), the vtx could not red the command. Not shure if the resistor is a pull down, but if yes I could imagin that it heps.

Unfortunatly i distroy my WING FC controller during measurment. Due to that I don't have pciturs.
I could also imagin that there migt be a SW controllible pull down which could solve the issue too.

activa73, 2020-05-06:
@datamann Hello, I have the same issue than you with myWING FC and smart audio on TX4. Have you try to change the resisitor from 1k to 100 ohm? And it is work?
@dangerg Hello can you confirm it is work, please?
BR

activa73, 2020-05-11:
I confirm Smart audio works after changing resitor from 1k to 100 ohm. The size is 0402.

With 1kohm : waveform uC side
[image]
With 1kohm : waveform TX4 pin side
[image]
With 100 ohm : waveform uC side
[image]
With 100 ohm : waveform TX4 pin side
[image]

modwadd, 2020-05-12:
Good to know, thanks for the effort to do the measurement.

dangerg, 2020-05-12:
Thanks for the great work.

activa73, 2020-05-12:
No problem thank you for the solution.

UnchartedBull, 2020-07-23:
Can confirm that this works with 150 Ohm as well. You can also use a „normal“ resistor and solder one end to tx4 and the other to the side of the 1k resistor that is closer to the CPU. Just in case you don‘t want to desolder this tiny resistor (electricity will always use the path of least resistance) plus it is easily reversible :). Thanks for the info guys!

[image]

hallskelet, 2022-07-08:
Спасибо вам огромное за описание этой проблемы. Производитель не даёт схему этой платы и без вас я бы ещё долго искал проблему.

andrushm, 2024-05-06:
Helpful even in 2024! Thanks!