iNavFlight/inav #4247: Switching Receiver Mode in flight

shlapion, 2019-01-22:
Greetings, 

was wondering if it would be possible to adapt iNav to be able to switch receiver modes in flight. 
I am working on a project where I want to control my quad using ROS. The communication between the FC and ROS happens via MSP. Everything works great however there is no way that I could take manual control of my quad in case my ROS nodes fail. So I was thinking of introducing a manual override function, which would allow me to switch between MSP and for ex. SBUS in flight without the need of a reboot.  The reason I would like to implemented on the FC is to have a decoupled system; in case my companion computer freezes mid flight would like to have the ability to fall back on the iNav functionalities.

Any suggestion is greatly appreciated

To conform to the suggested format:

## Current Behavior
Currently we need to reboot the FC to be able to switch form one mode to another. 

## Desired Behavior
Extend the functionality of iNav such that we can switch between two ore more receiver modes in flight.

## Suggested Solution
I was giving it some thought and in my opinion a theoretical way to tackle this issue is to allow a channel to be mapped to enable a switch between the two receiver modes. 
So for ex we could map channel X such that when it is in one position the Receiver mode is set to SBUS and when it is set to another position the receiver mode is set to MSP.  I am not sure if this is something which can be implemented in practice. Wanted to bounce the idea off the community. 

## Who does this impact? Who is this for?
Feature would be useful for iNav users who would like to experiment with autonomous flight and wish to have the ability to take manual control of their vehicle in case of failure of their experimental build.(software or hardware)

shellixyz, 2019-01-28:
See #4084

nik012003, 2019-01-28:
@shlapion and I are working on this feature.
We created a Combined receiver_type and added it to settings.yaml and rx.c/rx.h. Then we made combined.c/.h where it initalises sbus(or any other protocol) and MSP.
The code is on https://github.com/nik012003/inav

The problem is that the function sbusInit is called, receiver_type gets changed from COMBINED to SERIAL.
I'm looking around in the code to see why it does this but I still haven't found the cause.
Could you @shellixyz have a look?

shellixyz, 2019-01-29:
@nik012003 

You're changing it back to SERIAL in `combinedRxInit`

```
combinedRxInit(rxConfig(), &rxRuntimeConfig);
[...]
void combinedRxInit(const rxConfig_t *rxConfig, rxRuntimeConfig_t *rxRuntimeConfig)
{
    combinedRxConfig = rxConfig;
    combinedRxConfig->receiverType = RX_TYPE_SERIAL;
    combinedRxConfig->serialrx_provider = SERIALRX_SBUS;
```

shlapion, 2019-01-29:
@shellixyz  thank you for the insight. Will take a look at your suggestion. 
A small setback me and @nik012003 are facing is that our dev FC (MatekF722) is not able to receive SBUS input with iNav 2.0.1. 
For this we are currently working on separate versions of iNav namely 2.0.1 and 1.9.1.

shellixyz, 2019-01-29:
You're passing a pointer to the the `rxConfig` struct that contains the system RX config to your function `combinedRxInit`. Then you assign this pointer to `combinedRxConfig`. So when you are doing `combinedRxConfig->receiverType = RX_TYPE_SERIAL;` what you're doing is equivalent to doing `rxConfig()->receiverType = RX_TYPE_SERIAL;`, changing the receiver type back to serial.

I will take a look at the F722 S.Bus input issue.

shellixyz, 2019-01-29:
If you want a copy of `rxConfig` to modify then pass to `sbusInit` declare `combinedRxConfig` like this `rxConfig_t combinedRxConfig`. Then in `combinedRxInit` do `memcpy(&combinedRxInit, rxConfig, sizeof(combinedRxInit))` to copy the content of `rxConfig` in `combinedRxInit`. After that you can alter the content of `combinedRxConfig` and it won't change `rxConfig`. Then you can call `sbusInit(&combinedRxConfig , combinedRxRuntimeConf)`. It can be a bit tricky to understand pointers. They only point to stuff. If you just copy a pointer you don't copy the data the pointer points to.

shlapion, 2019-01-29:
If i am not mistaken, the solution you proposed in #4084  wold be a little more flexible. 
The implementation in @nik012003's version we were specifically targeting S.BUS, so maybe it would make sense to focus on a "universal" solution. 
Will have a talk with @nik012003 and see how best to take the dev of this feature forward.

nik012003, 2019-01-30:
Ok, so we followed the solution proposed in #4084 and tested it on a PPM receiver, everything is working fine. The code is on https://github.com/nik012003/inav/tree/1.9.1-override . 
I had to use 1.9.1 as base because of SBUS problems with newer versions.

On the CLI you can enable the feature by doing `set msp_override = ON` (It's ON by default right now)
You need to set up a MSP OVERRIDE mode on an aux channel by doing `aux 10 32 6 1000 1500`.
Now attach a MSP rx and another type of RX (serial or other) and you can see that when you go over 1500 on the selected aux channel, the first 6 channels are set by the normal RX, and if you go under 1500, the first 6 channels are set by the MSP RX

We still need to clean the code a bit, add the CLI option to change which channels must be overwritten by MSP and finally port it to development branch(after resolving the SBUS issues).

Thank you very much @shellixyz for the help!

teckel12, 2019-01-31:
@nik012003 MSP on SBUS was significantly reworked in about 1.9.1. I know because I did most of it. Getting your code to work with 2.1.0 will be required to merge it into the code base.

Good luck, seems like a novel idea. Kinda a unique situation, but valuable to some nonetheless.

shellixyz, 2019-01-31:
I just checked S.Bus with the Matek F722, working fine for me I can't reproduce your issue

shlapion, 2019-02-01:
Can confirm the override functionality works. 
@shellixzy will changed my recevier and will test it out with the 2.x versions of iNav.

closed 2019-02-01 by shlapion