iNavFlight/inav #4248: Personalization name of the model dissapears from 2.0 to 2.1.0 and 2.1.1  configurator for matek 405d

fmartinezo, 2019-01-22:
## This is a 2.0 screen
[image]
This is a 2.1.0 
[image]
This is a 2.1.1
[image]
 

## Steps to Reproduce
1. start configurator 
2. go to config page
3. scroll down

- FC Board name and vendor: matek 405 ctr
- INAV version string: # version
# INAV/MATEKF405 2.1.0 Jan 22 2019 / 09:41:21 (7bdd5967e)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

teckel12, 2019-01-22:
@fmartinezo I noticed this too, but figured it was just moved and didn't care enough to go looking.

fmartinezo, 2019-01-22:
I understand that you do not care too much, but from what I've looked at it has simply disappeared from the configurator. However, it can be accessed from cli.

fmartinezo, 2019-01-22:
@teckel12 I remind you that the inav telemetry lua parameterizes the models by the name and if people do not see where to put the name or do not know how to do it, they will not be able to parameterize the models. So it is very important that the place where the name is placed reappears in the configurator

teckel12, 2019-01-23:
@fmartinezo I think you misunderstood my message. I didn't mean that it shouldn't be fixed, I just meant I didn't have the time to go looking for it at that time so I just entered it in the CLI.

fmartinezo, 2019-01-23:
It is already open in configurator. You can close here

closed 2019-01-23 by fmartinezo