iNavFlight/inav #4255: SP F3 high cpu load

tomasssk, 2019-01-24:
Using the latest 2.0.1 with my SP F3 deluxe, I have constantly high CPU load (around 900%). The response in the Inav gui is very laggy and also the pre-arming check "cpu load" is red.

When I type "tasks" in the CLI, this is what I get:

# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9       7       4    0.5%    0.5%        12
 1 -          PID    1605     167     143   27.3%   23.4%     66197
 2 -         GYRO    1607     481     431   77.7%   69.7%    239450
 3 -          ACC     227      24      19    1.0%    0.9%      1558
 4 -     ATTITUDE      49      64      58    0.8%    0.7%      1005
 5 -           RX      24      70      60    0.6%    0.6%       566
 6 -       SERIAL      16     985      33    2.0%    0.5%       326
 7 -      BATTERY      24      19      14    0.5%    0.5%       122
 8 -  TEMPERATURE       0       5       3    0.5%    0.5%         0
 9 -       BEEPER      44      21       8    0.5%    0.5%       134
11 -      COMPASS       4     236     233    0.5%    0.5%       408
12 -         BARO      47     262     252    1.7%    1.6%      4219
19 -          CMS       8       6       4    0.5%    0.5%        13
21 -      VTXCTRL       0       0       0    0.5%    0.5%         0
Task check function            16      11                        94
Total (excluding SERIAL)                   112.6%  100.4%

I have this problem with the default settings, but also when I tried to apply different preset (racer) there is no change.

What could be wrong? I tried to flash both cleanflight and betaflight before and both were working without any issues...

tomasssk, 2019-01-24:
Also a status log...

# status
System Uptime: 21 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.0V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050, BARO=MS5611, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 72 MHz
  HCLK   = 72 MHz
  PCLK1  = 36 MHz
  PCLK2  = 72 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10002000, Heap available: 816
I2C Errors: 0, config size: 3509, max available config: 6144
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 863, cycle time: 1392, PID rate: 718, RX rate: 24, System rate: 9
Arming disabled flags: OVRLD RX CLI SETTINGFAIL
Invalid setting: accgain_x

The accelerometer/compass I have calibrated several times as required...

tomasssk, 2019-01-24:
Problem solved.
I chose Flight Controller Loop Time = 250 Hz
Now CPU load is around 4%

closed 2019-01-24 by tomasssk

giacomo892, 2019-01-24:
There is something wrong with your board / setup.
Invalid setting: accgain_x for instance..

250Hz is barely ok for a wing, will not work on a quad.