iNavFlight/inav #4259: Pre-arm "settings validated" red, incorrect acc_gain

tomasssk, 2019-01-24:
After calibrating the accelerometer, the "settings validated" flag is red and the status gives:

# status
System Uptime: 131 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 15.61V (4S battery - OK)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050, BARO=MS5611, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 72 MHz
  HCLK   = 72 MHz
  PCLK1  = 36 MHz
  PCLK2  = 72 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10002000, Heap available: 816
I2C Errors: 0, config size: 3509, max available config: 6144
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 9, cycle time: 2005, PID rate: 498, RX rate: 49, System rate: 9
Arming disabled flags: RX CLI SETTINGFAIL
Invalid setting: accgain_y

When I go to the calibration tab, the acc gain is 8186(x), 8226(y) and 7951(z).
In the "sensors" tab the output of the accelerometer looks perfectly fine.

giacomo892, 2019-01-24:
#defaults from CLI and do the setup again.

tomasssk, 2019-01-24:
No change. Went to #defaults, rerun the configuration and the issue moved now to accgain_x:

# status
System Uptime: 14 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.0V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050, BARO=MS5611, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 72 MHz
  HCLK   = 72 MHz
  PCLK1  = 36 MHz
  PCLK2  = 72 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10002000, Heap available: 852
I2C Errors: 0, config size: 3509, max available config: 6144
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 3, cycle time: 2016, PID rate: 496, RX rate: 49, System rate: 9
Arming disabled flags: RX CLI SETTINGFAIL
Invalid setting: accgain_x

giacomo892, 2019-01-25:
Do you mind upgrading to 2.1 and clean flash?

tomasssk, 2019-01-25:
Ok, reflashed to 2.1.0.
For the issue there is now just a code without a description...

# status
System Uptime: 20 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.0V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050, BARO=MS5611, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 72 MHz
  HCLK   = 72 MHz
  PCLK1  = 36 MHz
  PCLK2  = 72 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10002000, Heap available: 388
I2C Errors: 0, config size: 3537, max available config: 6144
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 11, cycle time: 1018, PID rate: 982, RX rate: 49, System rate: 9
Arming disabled flags: 0x4140000

tomasssk, 2019-01-25:
And this is how my calibration tab looks like

[image]

stronnag, 2019-01-25:
If you mean the arming flags, it is described:
https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

giacomo892, 2019-01-25:
Load the mixer and you should be fine...

stronnag, 2019-01-25:
The arming flags still include `SETTINGFAIL`. Either the OP fails to perform a full chip erase or the board is  broken.

tomasssk, 2019-01-25:
Not really...I loaded the mixer with the default Quad X but the issue is still the same

[image]

tomasssk, 2019-01-25:
> The arming flags still include `SETTINGFAIL`. Either the OP fails to perform a full chip erase or the board is broken.

The "full chip erase" I have selected when flashing the new hex file...so that should be ok...

stronnag, 2019-01-25:
In that case, I fear your board is physically damaged. SPRF3 works fine here.

tomasssk, 2019-01-25:
But the "settings validated" becomes red only after the accelerometer calibration.
Just after the flashing,it is green...

tomasssk, 2019-01-25:
Also, how come the verification of the content passes...in case the memory is faulty I would expect the verification to fail...

stronnag, 2019-01-25:
I rest my case ...

tomasssk, 2019-01-25:
Is this check different from cleanflight? Because there everything works without any issue...

stronnag, 2019-01-25:
Yes, as is the calibration. iNav much less not tolerant of bad hardware.

closed 2019-02-27 by DzikuVx

NiharX1997, 2019-05-30:
Hii @tomasssk, have you solved this issue....?? If not then plz tell me so i can solve it because i was exactly facing this issue on my spracing F3 FC with inav 2.1.0 firmware then after some research finally i fixed it.