iNavFlight/inav #4262: Target request: Mamba F405 FC

kn8alpha, 2019-01-25:
This looks like a good quality, rightly priced F4 AIO FC + 4-in-1 ESC stack.

https://www.diatoneusa.com/store/p587/MAMBA_F405_Betaflight_Flight_Controller_F40_40A_3-6S_DSHOT600_FPV_Racing_Brushless_ESC.html

Except for baro, has everything else one would need for a basic iNav setup.. OSD, 2xUARTs, I2C pads.

Betaflight supports this using Fury F4 OSD ( FURYF4OSD )

chrisogg, 2019-02-08:
Yes, NAZA GPS / Mag appears to function well at least in terms of sat links.
All I have are BPM180 external baro's and none of the FURYF4 or FURYf4OSD targets have this included.
Good reason for me to finally get the dev env set up.

chrisogg, 2019-02-08:
Oh and the ~$40 stack performance has been excellent so far.
Light weight and quality construction.
Running it in a light weight sloss 7" 5/6S LR set-up which would be totally awesome with INAV!

Dronitoni, 2019-02-09:
my Mamba F405 FC has been delivered today.
So, I'm waiting for a INAV target :-)

dentex, 2019-02-10:
Hello,
I managed to put together the build files for this target... but the majority of the stuff it's just guessed.
https://github.com/dentex/inav/tree/dentex_add-furyf4osd-target/src/main/target/FURYF4OSD
I basically had a look at what they did in Betaflight and I had the `target.c` file from a user on the Telegram chat (S.Prokofev) who gave me an iNav version for this FC that didn't work (it shoots to the moon as soon as you open the throttle).
I disabled (...I think, at least...) Dshot, commenting the relevant row in `target.c` (maybe the problem was just that).
I also added some details to use the Led strip (the guessed part mentioned above).

If you want to take a look... That would be great... Thanks.

It goes without saying... Don't flash this build.
I'll do some tests myself ASAP.

kn8alpha, 2019-02-17:
I've made a target for the MAMBA F405 using inav 2.1 RC2 and tested on my quad with the following HW/features:

X6B ibus receiver
BMP280 baro
BN880 GPS / USE_MAG_HMC5883 compass
Smartaudio to TX805 VTX
OSD, Blackbox

I also tested with SBUS (XM+) receiver but getting erratic numbers on the receiver tab. The same RX works with BF with the FuryF4OSD target, so this is an iNav/target problem.

Attached HEX file has iBus default.

[dump mamba f405.txt](https://github.com/iNavFlight/inav/files/2872996/dump.mamba.f405.txt)

[inav_2.1.0_MAMBAF405OSD hex.zip](https://github.com/iNavFlight/inav/files/2872994/inav_2.1.0_MAMBAF405OSD.hex.zip)

[MAMBAF405.zip](https://github.com/iNavFlight/inav/files/2872988/MAMBAF405.zip)

dentex, 2019-02-17:
Hello @kn8alpha 
I'll try your build ASAP (I have a PPM receiver on that quad), the same (external) barometer and a similar GPS/MAG module.
I'll let you know.

kn8alpha, 2019-02-17:
I had missed including the UART inverter related macros in target.h
Now SBUS works too, with default CLI parameters.

[MAMBAF405.zip](https://github.com/iNavFlight/inav/files/2873238/MAMBAF405.zip)

[inav_2.1.0_MAMBAF405OSD (2) hex.zip](https://github.com/iNavFlight/inav/files/2873240/inav_2.1.0_MAMBAF405OSD.2.hex.zip)

I haven't tested LED strip and buzzer yet.

DzikuVx, 2019-02-18:
@kn8alpha please make it a pull request if you want ever to be included in INAV

dentex, 2019-02-18:
@kn8alpha All seems to be working OK. I'll go out to test it later today.
Just one thing: the `WS2811_PIN` is `PA0`
:)

michaelturibio, 2019-03-18:
@ kn8alpha As you turned on the GPS bn880 and the bn280 barometer together. The compass uses the i2c ports. Where does the barometer weld?

netmute, 2019-03-18:
@michaelturibio I2C is a multi-device bus. You can connect multiple devices in parallel on the same bus. The controller will use the different I2C device addresses to figure out what's what.

sergeyprokofev, 2019-03-19:
Current and rssi, works correctly if, to change them.
#define USE_ADC
#define ADC_INSTANCE                ADC1
#define ADC1_DMA_STREAM             DMA2_Stream0
#define ADC_CHANNEL_1_PIN           PC1
#define ADC_CHANNEL_2_PIN           **PC3**
#define ADC_CHANNEL_3_PIN           **PC2**
#define VBAT_ADC_CHANNEL            ADC_CHN_1
#define CURRENT_METER_ADC_CHANNEL   ADC_CHN_2
#define RSSI_ADC_CHANNEL            ADC_CHN_3

shellixyz, 2019-03-19:
@kn8alpha I made a PR with your target #4521

gsixx8, 2019-04-25:
@kn8alpha I am running this on my Mamba FC as well.  The LED functions dont work properly.  I just get white LED's even tho I have set specific colors in the LED tab of the configurator.  

Buzzer works fine (using a VIFly buzzer).

I am getting the "NOT OK" message in the lua script.  Even after I have over the minimum GPS sats.  I can arm and fly, but the "NOT OK" message remains.  I set nav_extra_arming_safety = on (i was instructed to do this by another user here to possibly deal with the "NOT OK" message)  

I am also being told that the iteration  of 2.1.0 inav you used to make this compiling was and RC and  not the stable finished 2.1.0 version.  All the Crossfire stuff was not finalized in the RC.  Any way you could compile this using the stable and complete 2.1.0 version?

PitrisN, 2019-04-29:
Hello Would I ask where did you solder RSSI signal? I´m trying to find some information but I got info from Diatone that there is no pad for analog RSSI. I´m thinking about soldering it directly on MCU pin (over RC filter because of RSSI is PWM). Thank you

gsixx8, 2019-04-30:
@kn8alpha would you compile the MambaF405 target again using the current stable version of iNav?

playermty, 2019-05-07:
works with mamba f405 mini?

DzikuVx, 2019-05-09:
Since traget is there, we can close it I think

closed 2019-05-09 by DzikuVx

kekek95, 2019-06-18:
Hi,
i started building a fixed wing with this FC, but it doesnt have any servo outputs. Can someone add them to the target?

SECOUM, 2019-06-23:
Hello all,
I use your Zip file for mamba f405 ,all works good

Today i flash with official mambaf405 2.2 and impossible to connect rxsr with sbus!???
Do you know what is the problem ??? 
Thancks

arvi-kris-cy, 2019-06-29:
Hi @SECOUM,

I think the issue might be because UART inverter is not enabled in the official target.
Try adding these to the official target.h (I don't know if this would conflict with FuryF4OSD UART1 behavior)

#define USE_UART_INVERTER
#define UART1_AHB1_PERIPHERALS  RCC_AHB1Periph_DMA2
#define INVERTER_PIN_UART1_RX   PC0 // PC0 used as inverter select GPIO

pavel4441, 2019-07-11:
Hello,

With both the latest official one and the one here, BMP180 (assume same as BMP085) still doesn't work, stays red. The gps is at UART6, works, as everything else.

datamann, 2019-08-05:
I just tried to build mambaf405 from development and SBUS and softserial on PPM pad does not work. I tried the build (zip file above) and it works but that build is from an older code base. Did also try FuryF4OSD and SBUS does not work with that one either.

nandakoryaaa, 2019-09-20:
On the Mamba F405 Mini, target MambaF405 with the latest Inav version (2.2.x) does not have any of the UARTS working. The build that is linked above in this thread (inav_2.1.0_MAMBAF405OSD (2).hex) works fine.

naice, 2019-10-10:
same here, on inav_2.1.0_MAMBAF405OSD (2).hex the mixer config is trash and on the latest release sbus is not working :/

luudatdungVN, 2019-12-20:
can i use RTH with this Fc. i have GPS M8N. but this Fc dont have Baro . my Fc is mamba F405

JasonFPV0, 2019-12-26:
So as of right now, I would not be able to use sbus with this? It would be great if this was completed, but since it was started nearly a year ago I don't think it will be finished.

quetiano, 2020-04-07:
Good day all.
Has anybody ended up using the ms5611 baro on this FC in inav?
I am new and since inav doesnt support remapping like betaflight, I cant get the this baro to work in inav 2.4. 

I want to remap the uart 3 to i2c_2 as this worked in betaflight. My mamba f405 mk2 ran this baro with the compass in betaflight, however, when connecting the compass and baro to the normal I2C soldering pads, they did not work. According to the pinout diagrams, I saw that the I2C_2 ports existed on the uart 3, so then I connected the compass to the normal I2C_1 port and the baro to the I2C_2 ports, did some cli configs in betaflight, and I got all sensors to work.

In inav, since theres no remapping, I need that I2C_2 ports, so can anyone create a hex flashable file with the uart 3 remapped to I2C_2 ports for this FC ? Or even teach me how to and I can do it myself, because when I tried to extract the firmware from the mamba using simpleIDE, simpleIDE always failes with an error, and mamba lights up red.

Narnasqueneth, 2020-05-04:
> works with mamba f405 mini?

Anyone get a response to this? I've got a couple f405 mini's I could put to good use...

nandakoryaaa, 2020-05-04:
I reflashed my mamba mini with the newest inav 2.4 and everything works.

Narnasqueneth, 2020-05-05:
Thanks but which target did you use as I cannot see f405MINI appearing in the list of targets?

all the best!

nandakoryaaa, 2020-05-05:
The target is MAMBAF405, it works for mini also

lavachemist, 2020-10-20:
Does the MAMBAF405 target work for the MK2 version of the board as well?

nandakoryaaa, 2020-10-20:
> Does the MAMBAF405 target work for the MK2 version of the board as well?
Should work.

alexvazquez, 2020-11-26:
> Does the MAMBAF405 target work for the MK2 version of the board as well?

I have MK2 did you try it?

alexvazquez, 2020-11-26:
Hi, Have a Mamba F405 MK2 that want to use for a wing or airplane. Pins 1,2,3,4 are the ones used for Motor and servos? Any special wiring that I need to do? Appreciate so much any help

lavachemist, 2020-11-26:
> Hi, Have a Mamba F405 MK2 that want to use for a wing or airplane. Pins 1,2,3,4 are the ones used for Motor and servos? Any special wiring that I need to do? Appreciate so much any help

The firmware is only configured with multirotor motors, but you do have two timers, so you should be able to use this board on a flying wing with two servos and up to two motors. You'll have to flash the firmware and see how the mixer sets it up, but I would assume it will put motors on S1-S2 and the servos on S3-4. If you aren't using LEDs you can create custom firmware and have a third servo by remapping the LED pad.

lavachemist, 2020-11-26:
> > Does the MAMBAF405 target work for the MK2 version of the board as well?
> 
> I have MK2 did you try it?

Yes, the firmware works. I'm using it on a rover.

stronnag, 2020-11-26:
> The firmware is only configured with multirotor motors, but you do have two timers, so you should be able to use this board on a flying wing with two servos and up to two motors. You'll have to [...]

#6165

lavachemist, 2020-11-26:
> #6165

Oh, nice, I must have an older version of the target. Either way, everything else I said holds true. It will put the motors on S1-S2 and the servos for S3-4, and if you need another servo, you'll need to remap the LED pad and compile your own firmware.

alexvazquez, 2020-11-26:
> #6165

Awesome, got it. How difficult is to create a custom firmware so I can use the LED's pin?

On the other hand, how does this work for a quad? I havent use INAV for quadcopter but I have a Tarot Ironman 650 frame with motors and ESC's that I was using with Mini APM 3.1 long time ago. How does INAV with MambaF405 MK2 would work for this quad? What do you think?

stronnag, 2020-11-26:
> Awesome, got it. How difficult is to create a custom firmware so I can use the LED's pin?

Depending on one's software development skills, varies from trivially easy to impossible. 
 
> On the other hand, how does this work for a quad? I havent use INAV for quadcopter but I have a Tarot Ironman 650 frame with motors and ESC's that I was using with Mini APM 3.1 long time ago. How does INAV with MambaF405 MK2 would work for this quad? What do you think?

A quad still has 4 motors, that's what MC (multicopter) and FW (fixed wing) mnemonics indicate in the `target.c` macros.

alexvazquez, 2020-11-26:
Appreciate so much. Will try to setup for Quadcopter. Seems little complex to do the custom compiling for adding an extra UART's for the LED I guess. Hopefully someone already did it

Narnasqueneth, 2020-11-28:
> Does the MAMBAF405 target work for the MK2 version of the board as well?

Yes it does

Locfpv, 2021-07-24:
anyone help me , what is the mamba f405 mk2k firmware target configuration ? Pls

Locfpv, 2021-07-24:
anyone help me , inav what is the mamba f405 mk2k firmware target configuration ?

PR #4521 (2019-03-26): Add MAMBAF405 target
Closes #4262