iNavFlight/inav #4264: Add Kakute F7 AIO

d4nieel, 2019-01-25:
Hello, I am Daniel and I am new here.
I have been running a flip32 FC on a drone with a little gimbal and the next step is to add GPS to it. I bought a Kakute F7 AIO and put everything together and installed the software mentioned on the target [#3868](https://github.com/iNavFlight/inav/pull/3868), this one: [inav_2.1.0_KAKUTEF7.hex.zip](https://github.com/iNavFlight/inav/files/2676704/inav_2.1.0_KAKUTEF7.hex.zip). The problem is that as it is made for the Kakute F7 (standard version) this firmware doesn't support the OSD that comes on the AIO version. I would like to add this function and then (just for me) I would like to change the pin from the motor 5 (C09) as servo 1 to control the gimbal pitch as I have been doing till now. I have seen this  [Paweł Spychalski's video](https://www.youtube.com/watch?v=v4R-pnO4srU) and I think that timers shouldn't be a problem. I would like to learn about all of this and create my own custom firmware but I have seen that [the guide](https://github.com/iNavFlight/inav/wiki/Making-a-new-Virtualbox-to-make-your-own-INAV) is obsolete as Arch-anywhere is not longer avaliable so I am planning to use an other distribution (Any suggestions?).  I am studying systems engineering and I have a base on C++ programming  but I don't know if it will be difficult so I need some advice. I don't want you to do it for me, I want to learn to be able to help in the future.
Thanks and regards.

geniusboywonder2020, 2020-03-24:
@d4nieel Did you fix this? i am trying the same thing except i have the non AIO version and cannot get that hex file to work.

d4nieel, 2020-03-26:
Hi @geniusboywonder2020 ,
I ended buying a compatible board, if the problem has not been solved yet I don't think it can be worth.

geniusboywonder2020, 2020-03-26:
> Hi @geniusboywonder2020 ,
> I ended buying a compatible board, if the problem has not been solved yet I don't think it can be worth.

Hi, what board did you buy and do you have 4 motors and 2 servos running that is controllable