iNavFlight/inav #4267: some CLI parameters not safed

wilco1967, 2019-01-25:
## Current Behavior
Inav 2.1 RC2
Matek405SE Wing

The following parameters are not saved in the CLI. The always revert back to default.
may be more, but these showed up when comparing.

set osd_artificial_horizon_max_pitch = 30
set osd_crosshairs_style = AIRCRAFT
set osd_left_sidebar_scroll = GROUND_SPEED
set osd_right_sidebar_scroll = ALTITUDE
set osd_sidebar_scroll_arrows = ON

set osd_failsafe_switch_layout = OFF

## Steps to Reproduce

1. set the parameters in CLI
2. save
3. dump
4. compare dump with what you tried to load.

## Expected behavior

## Suggested solution(s)
guess some parameters where forgotten... 

## Additional context

---

# INAV/MATEKF405SE 2.1.0 Jan 22 2019 / 09:42:54 (7bdd5967e)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

shellixyz, 2019-01-29:
Please test with the latest released version of the configurator

https://github.com/iNavFlight/inav-configurator/releases/tag/2.1.1

wilco1967, 2019-01-29:
Hey, it works.....
how come.... ?

I'm pretty sure I used the 2.1.1 configutator before, and tried a few times to save those parameters, and no good... Even checked the version 2.1.1. I had installed.
just to exclude any screwups, I downloaded the configurator again, exact same file... deleted the entire iNav-configurator 2.1.1 directory, and re-installed.
Now it takes it ok.... 

Strange thing is (or was), another Matek F722 took those parameters OK already before, but the 405SE Wing did not....
but now it does....

Don't understand where I screwed up, but somehow I probably did....

Even more strange is, I uploaded through CLI. So version shouldn't matter at all, or does it ?

never mind.... problem gone.... 
closing.

Thanks for your help !

closed 2019-01-29 by wilco1967