iNavFlight/inav #4268: Incorrect position of fixed OSD elements in PAL mode

Arvycka, 2019-01-26:
## Current Behavior

fixed OSD elements - AHI, Crosshair, sidebars are shown one line higher in inav 2.1-RC2 OSD PAL mode compared to Inav 2.0 

## Steps to Reproduce
1. Open OSD tab in configurator and setup a layout in PAL mode.
2. Compare to the actual layout on the screen - Croshair position on the screen does not match position in the configurator preview:

[image]

## Expected behavior
 Fixed OSD elements in PAL mode are supposed to move down by one line in PAL mode compared to NTSC mode:

[image]

## Suggested solution(s)
It would appear that video format change to PAL is ignored regarding the position of the fixed OSD elements.

---
INAV/MATEKF405SE 2.1.0 Jan 26 2019 / 13:05:51 (506c64f)
GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

Arvycka, 2019-01-28:
Seems to have been fixed.

closed 2019-01-28 by Arvycka

fiam, 2019-01-28:
@Arvycka Can you test the patch? I can build a binary for you if needed.

Arvycka, 2019-01-28:
I just did - Both PAL and NTSC. It's as was before 2.1. AHI block moves to line 7 (8th line) for PAL as is drawn on line 6 for NTSC.

fiam, 2019-01-28:
Thanks, perfect!

PR #4284 (2019-01-28): Check the correct bit in VM0 to detect screen size
Check VM0 against VIDEO_MODE_PAL (bit 6 = 1). Previous check was
a typo that always returned true.

Fixes #4268
commit d6a8dd6e (2019-01-28): Check the correct bit in VM0 to detect screen size

Check VM0 against VIDEO_MODE_PAL (bit 6 = 1). Previous check was
a typo that always returned true.

Fixes #4268