iNavFlight/inav #4269: OmnibusF4V3 connect a laser Rangefinder VL53L0X ,it will crash ( INAV 2.1.0-RC2)

Missionhou, 2019-01-26:
## Current Behavior

Connect VL53L0X to OminbusF4V3 board throuth I2C bus, if OminbusF4V3 board flashed INAV 2.1.0-RC2,it can not bring up the board, flash back to INAV 2.0.1,it working normal
## Steps to Reproduce

1.Connect VL53L0X board to OmnibusF4V3 board through I2C bus
2.OmnibusF4V3 board flashed INAV 2.1.0-RC
3.Try to connect OmnibusF4V3 to INAV Configurator2.1.1,Unable to connect
4.flash back to  INAV 2.1.0,it working normal

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Missionhou, 2019-01-26:
3.1 first time  it  can connect to INAV Configurator2.1.1,then go to configuration page to select Rangefinder as VL53L0X and save,And then you can't connect again

digitalentity, 2019-01-26:
Interesting. Just tested - all working normally on OmnibusF4 PRO board, so this is either board-specific or setup-specific.

Missionhou, 2019-01-27:
I also try OmnibusF4Pro board flashed inav_2.1.0 Version ,it still can not bring up(led flashes 3 times and then stop flashes) ,if I remove VL53L0X I2C wires,it return to normal,and in configurator sonar icon became red. I flashed back to inav_2.0.1,  connect I2c wires, sonar icon come back blue.

digitalentity, 2019-01-27:
Bug confirmed. Not sure about the cause though.

digitalentity, 2019-01-27:
Root cause fixed by #4249. Further prevention in #4276

closed 2019-01-27 by digitalentity