iNavFlight/inav #4271: Matek f405ctr GPS issues

Ferris75, 2019-01-26:
Hi all, new to this hobby and have a matek f405 ctr with led on pin s5 but cannot connect GPS. Have GPS added as per the matek website image however mag and baro working but GPS in red in Inav. Wondering if I have a configuration issue due to on board baro on the fc board. So frustrated that there's nothing clear out there to help on this, anyone got any suggestions on what to do. I can screenshot ports (GPS on uart3) baro and mag work g GPS has fix on location but showing as red in inav. Tha ks in advance.

laurentopia, 2019-01-26:
fix type = 3D?
in case you're missing something, my setting for the BN880 is
in ports : UART4 sensors = GPS
in configuration : GPS = on, protocol = UBLOX, ground assist = Autodetect
*connect battery and marvel at the blue GPS

Ferris75, 2019-01-27:
That doesn't work either. I think I have issues with my port settings or similar but just can't work it out. There's no wiring diagrams out there with comprehensive details on the matek board.

Does someone have any experience e with the f405ctr and a radiolink se100 and inav showing as red.

Tha ks for the reply lautentopia but not sure what you mean. I had GPS on uart 3 set up for GPS but just no GPS fix and not showing in GPS tad of inav.

Ferris75, 2019-01-27:
Hi Laurentopia, so i have changed this to Uart4 and yes i now have a gps fix showing in GPS tab of Inav. Only issue i have now is that the GPS does not seem to be showing any led's other than power on (1 green led) but 3d frix in GPS... well confused now!!
[image]

shellixyz, 2019-01-29:
Check that both the TX and RX wires from your GPS are connected to the chosen UART. GPS RX -> FC TX and GPS TX -> FC RX. Then all you should need to do is chose GPS in the sensors drop-down list of the ports tab for the chosen UART. Please post a file containing a dump of your configuration

Ferris75, 2019-01-29:
Hi shellixyz, got it working now on uart 4 and getting a GPS fix quite easily. I now have another issue in that I have bad oscillation when trying to land. Do I have to reduce the pid levels for GPS or normal pitch/roll etc and are there any guides to help with that? Sorry for all the questions first build..

shellixyz, 2019-02-02:
@Ferris75 There is no guide for that. What you can do is open your blackbox log with BB viewer to check what is causing the oscillations. The output of all PID controllers are now present in the logs. Attitude PID outpus are `axisP`, `axisI` and `axisD`. Navigation PID outputs are `mcPosAxis*` and `mcVelAxis*`.

closed 2019-03-01 by shellixyz