iNavFlight/inav #4272: OSD page is blank

laurentopia, 2019-01-26:
## Current Behavior
OSD tab is blank, only documentation button

## Steps to Reproduce
connect f405-ctr
set it up so all sensors are blue

## Expected behavior
OSD shows up in the OSD page

---
- FC Board name and vendor:
f405-ctr, matek
- INAV version string:
# INAV/MATEKF405 2.1.0 Jan 22 2019 / 09:41:21 (7bdd5967e)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
`# set *
Current settings:
looptime = 500 0 9000
gyro_sync = ON
align_gyro = DEFAULT
gyro_hardware_lpf = 256HZ
gyro_lpf_hz = 90 0 200
moron_threshold = 32 0 128
gyro_notch1_hz = 0 0 500
gyro_notch1_cutoff = 1 1 500
gyro_notch2_hz = 0 0 500
gyro_notch2_cutoff = 1 1 500
gyro_stage2_lowpass_hz = 175 0 500
vbat_adc_channel = 1 0 4
rssi_adc_channel = 3 0 4
current_adc_channel = 2 0 4
airspeed_adc_channel = 0 0 4
acc_notch_hz = 0 0 255
acc_notch_cutoff = 1 1 255
align_acc = DEFAULT
acc_hardware = MPU6000
acc_lpf_hz = 15 0 200
acczero_x = 66 -32768 32767
acczero_y = -19 -32768 32767
acczero_z = -199 -32768 32767
accgain_x = 4119 1 8192
accgain_y = 4076 1 8192
accgain_z = 4016 1 8192
rangefinder_hardware = NONE
rangefinder_median_filter = OFF
opflow_hardware = NONE
opflow_scale =  1.000  0.000  10000.001
align_opflow = CW0FLIP
align_mag = CW270FLIP
mag_hardware = HMC5883
mag_declination = 0 -18000 18000
magzero_x = -2 -32768 32767
magzero_y = -49 -32768 32767
magzero_z = 55 -32768 32767
mag_calibration_time = 30 30 120
align_mag_roll = 0 -1800 3600
align_mag_pitch = 0 -1800 3600
align_mag_yaw = 0 -1800 3600
baro_hardware = BMP280
baro_median_filter = ON
pitot_hardware = NONE
pitot_lpf_milli_hz = 350 0 10000
pitot_scale =  1.000  0.000  100.000
receiver_type = SERIAL
min_check = 1100 1000 2000
max_check = 1900 1000 2000
rssi_channel = 0 0 18
rssi_min = 0 0 100
rssi_max = 100 0 100
[... 284 lines cut ...]
osd_main_voltage_decimals = 1 1 2
osd_coordinate_digits = 9 8 11
osd_estimations_wind_compensation = ON
osd_failsafe_switch_layout = OFF
osd_plus_code_digits = 11 10 13
i2c_speed = 400KHZ
debug_mode = NONE
throttle_tilt_comp_str = 0 0 100
input_filtering_mode = OFF
name = 
mode_range_logic_operator = OR
stats = OFF
stats_total_time = 0 0 2147483647
stats_total_dist = 0 0 2147483647
stats_total_energy = 0 0 2147483647
tz_offset = 0 -1440 1440
tz_automatic_dst = OFF
display_force_sw_blink = OFF
vtx_halfduplex = ON
vtx_band = 4 0 5
vtx_channel = 1 1 8
vtx_power = 1 1 5
vtx_low_power_disarm = OFF
vtx_freq = 5740 0 5999
vtx_pit_mode_freq = 0 0 5999`

laurentopia, 2019-01-27:
it's a configurator 2.1.1 bug, solution: revert back to 2.1.0
2.1.0 is also faster, 2.1.1 shoulnd't have been released

a1axx, 2019-01-27:
Same have this on any firmware 2.*

Matek f405 Wing & kakute f4 v2

laurentopia, 2019-01-27:
It's not a firmware issue, it's a configurator bug. Downgrade to 2.1