iNavFlight/inav #4283: SBUS / PPM OUTPUT from FC

robthomson, 2019-01-28:
Not sure if INAV supports this but could be usefull.

Typically inav boards will happily give you around pwm servo drivers.

Unfortunately some models require way more.  If I use a head tracker; I need at-least 9.

If INAV could be configured to output an sbus signal or ppm signal on one of its pwm pins; you would have a very simple solution to simply plug in something like an RMILEC 18 Channel High Precision PWM/PPM/SBUS Signal Converter  to access the extended channels.

This would then still allow you to use something like CSRF input with all the telemetry.

I know its possible to use a PCA9685; however I cant help but feel that  many people would find a simple sbus translation of the data to be easier to use.

DzikuVx, 2019-07-25:
@robthomson it is possible to connect 2 SBUS devices in parallel from one receiver. The same applies to PPM signal. Since they are one-direction only, you can have multiple listeners connected
SBUS output on PWM pin is rather na no-go since it requires serial port.

closed 2019-07-25 by DzikuVx

robthomson, 2019-07-25:
Totally aware of that.

The request was more about outputting sbus from fc, when the inbound protocol is csrf