iNavFlight/inav #4287: pixel osd instead of char mode - would be nice, but is it possible in iNav?

Pairan, 2019-01-29:
Hi there,

came across a cheap OSD FC which not is of real interest. But what got me was, that it had a pixel osd:

[image]

is that something we will have to keep dreaming of or could that be a nice FUTURE update?

wx4cb, 2019-01-29:
love it, but i beleive that would depend on the driver chip for the OSD

giacomo892, 2019-01-29:
Just out of curiosity. What's the name of that FC?

shellixyz, 2019-01-29:
That doesn't look like an analog PAL/NTSC picture

Pairan, 2019-01-29:
> That doesn't look like an analog PAL/NTSC picture

agreed ... could be a montage! But I assume it to be a mix of MFD and Vector ... however, that doesn't matter. I guess its more about, can the char osd be ported to a pixel osd

giacomo892, 2019-01-29:
@Pairan thanks, and no. we are limited to this with the current OSD chip :)

Pairan, 2019-01-29:
uh, that's a sad message ... how comes? has nobody done this by now?

teckel12, 2019-01-29:
@Pairan Mainly because this is an analog hobby and everything being developed these days is HD and modern.  Companies are developing where the money is, and that's not with analog video technology.

teckel12, 2019-01-29:
@Pairan Here's an actual output of that OSD:

[image]

fiam, 2019-01-29:
@Pairan I had never heard of that FC. Could you post a link to it so we can take a look?

Coincidentally, a few hours ago I was discussing with someone if we should take a crack at writing an OSD with pixel addressing using just an STM32 and a few extra components.

gbine, 2019-01-29:
Not sure if this is worth mentioning, but ...
Anyone know what happened to PlayUAVOSD (https://github.com/PlayUAV)? Looks like the project is dead.
If I recall, they had a nicer high resolution, STM32 based, OSD that worked with MavLink and a couple of others.

Pairan, 2019-01-30:
> 
> 
> @Pairan I had never heard of that FC. Could you post a link to it so we can take a look?
> 
> Coincidentally, a few hours ago I was discussing with someone if we should take a crack at writing an OSD with pixel addressing using just an STM32 and a few extra components.

sure ... just head over to BG and search for "pixel osd" ... should be just one hit!

I like the thing, that you've already been discussing about it :) muy bien as you might would say, verdad?

shellixyz, 2019-01-30:
https://www.banggood.com/SN-L-Fixed-Wing-Flight-Controller-With-Pixel-OSD-AAT-Support-PPM-SBUS-RSSI-For-Mini-RC-Airplane-p-1419662.html?rmmds=search&cur_warehouse=CN

teckel12, 2019-01-31:
@Pairan To toot my own horn, you can do pixel-based attitude displays currently with INAV, but from your transmitter. Supports all FrSky transmitters and SBUS and Crossfire telemetry:
 https://github.com/iNavFlight/LuaTelemetry/blob/development/assets/iNavHorus.png

Pairan, 2019-01-31:
> 
> 
> @Pairan To toot my own horn, you can do pixel-based attitude displays currently with INAV, but from your transmitter. Supports all FrSky transmitters and SBUS and Crossfire telemetry:
> 
> [image]
> 
> https://github.com/iNavFlight/LuaTelemetry/

This is very cool - ... but I won't see that while flying FPV :)

teckel12, 2019-01-31:
@Pairan But who needs FPV when you can fly virtually from your transmitter?  When I add virtual gates to the attitude display, you won't need FPV anymore. 😉

Pairan, 2019-02-01:
Oh yeah .. sure ... pretty cool idea ... ;) lets turn the Taranis into Nintendo :D ... no, please, let me have my daily dose of FPV so I can manage the insanity in this world !

fiam, 2019-02-19:
I’m adding support for this for 2.2

mistyk, 2019-06-09:
Have you got info about this OSD chipset ? maybe we can help to make one Opensource

danarrib, 2019-07-26:
Following

Pairan, 2019-09-03:
> 
> 
> Have you got info about this OSD chipset ? maybe we can help to make one Opensource (fc board with pixel osd )

I just went through the product description and the PDF (Manual) ... but my Chinese is fugly and I didnt find any clues so far.

Pairan, 2019-09-03:
Did some more digging. Turned out, that there is a thread on RCG. I've been trough all the pages but no clues on the chip. So I kindly asked ...

https://www.rcgroups.com/forums/showpost.php?p=42658777&postcount=157

mistyk, 2019-09-03:
@fiam  are you working on it ? could be intersted to work together to build hardware for that !
don't hesitate to contact me ;)

Pairan, 2019-09-04:
some more hints on the components of the FC:
> Hi PaiRaN
> 
> There appear to be four IC's on the PCB. I've taken some photos (see attached). Even under my magnifier I'm struggling to ID them.

1 ARM STM32F405
2 44AH LM1881M ([Possibly a video sync separator](https://www.components-store.com/datasheets/2c/LM1881N-NOPB.pdf)?)
3 P15V 330QE B2 G5GC ([Possibly a quad 2 channel mux/demux](https://www.diodes.com/assets/Datasheets/PI5V330.pdf)?)
4 Unable to ID.
 
> I assume IC2 and IC3 have something to with the AAT and/or OSD. Regardless is looks very impressive.
> 
> Cheers
> 
> Jase 

[image]
[image]

@fiam  ... perhaps this helps you getting a clue on whats used

mistyk, 2019-09-04:
It shouldn't be a problem to duplicate this FC, most difficult part is to
add this feature on inav ! :)

Pairan, 2019-09-04:
:) ... I don't want to duplicate that FC :) Just add a pixel-based OSD as an alternative to the old OSD chip.
And to make it easier for the developers, I collect all the information needed and possible on that chip as it apparently works well with a STMF405.

Pairan, 2019-09-17:
# Other pixel based solutions : tinyOSD
Another option for a pixel OSD comes actually from germany by a guy calles Fishpepper or Simon Schulz.

He made a full graphic OSD and got it running on a STMF0 (if I recall it right). 
- He attached Betaflight to it
- He has a full open source repository here 
- it should be 
I'll leave the description to it here, so any interested could dig a bit deeper:

https://fishpepper.de/2019/03/11/tinyosd-tinyfinity-a-tiny-opensource-video-tx-with-full-graphic-osd/

## He states in the page:
> As mentioned before this project is based on two parts: Hard- and Software. The hardware details can be found in my tinyFINITY github repository:
> https://github.com/fishpepper/tinyFINITY
>I also uploaded the latest PCB revision to OSHPark, you can find this revision 0.3 here. The stepdown regulator part of this design was not tested yet as I powered it externally during the last testing phases.
>In order to make this board do something you will need the tinyOSD firmware. Lucky for you (and all chinese copycats) this is also available for free on my github repository:
>https://github.com/fishpepper/tinyOSD

MFlghts, 2019-09-27:
What is FrSky's pixel based OSD? Where is it used?

teckel12, 2019-09-27:
@MFlghts Sounds like a not yet released product. My guess is that we'll be hearing about it soon. FrSky had to get it working in INAV and Betaflight before release.  This is how new product releases should work. Not the other way around like with every flight controller or video transmitter.

Pairan, 2019-09-27:
look like that @teckel12

DzikuVx, 2019-12-24:
I suppose we can close it as FrSky Pixel OSD support is part of INAV 2.3 already

closed 2019-12-24 by DzikuVx

MenooMeow, 2025-01-30:
I would LOVE to have this capability (the first image).  I fly jets in real life and I'm just getting into RC.  All the new technology is daunting, to say the least.  I'm not really in love with flying an airplane like a dog on a leash in 3rd person.  I want to be able to fly it like a real aircraft.  That first image is exactly what is needed.  

- Airspeed with tape
- Altimeter with tape
- Pitch ladder with major divisions at 5 degrees, major divisions at 10 degrees, with appropriate symbology for pitch or bank angles in excess.
- A velocity vector would be really great if it was accurate.
- Eventually some kind of computer 3 degree glidepath to a user defined runway with a user defined course... kinda like an ILS but made purely from GPS (GNSS).
- Compass heading tape across the top with a digital heading.
- A "tadpole" showing the direction to the next waypoint or home.  (reference F-16 HUDs)
- A high 60 Hz+ refresh rate with smoothing.  This is important.  Instruments that refresh as slowly as some of the OSDs I've seen are not good for flying.

I saw this video:  https://www.youtube.com/watch?v=IDlBqk8QX1Y
Demonstrating the FrSky Pixel OSD.  It looks okay but the resolution looks like a Commodore 64.  Can this be replicated using the SpeedyBee F405 OSD chip at a higher resolution and higher refresh rate?

sensei-hacker, 2025-01-30:
@MenooMeow  Probably the closest you could get to all that you're asking for would be to build on OpenHD set up that it to your liking. Yeah I did say "build" - it's not plug and play, just take it out of the box and go.

BUT nothing comes for free.
Suppose you have the bandwidth to send X bytes per second. You can use those bytes however you want - To double the refresh rate, you can cut the resolution in half.

Or, you can increase the bandwidth by cutting the distance/range by 75%.

You want to draw 10X more pixels, to draw a ladder?
Sure, drawing pixels takes time and space, so you're going to pay in both dollars for a higher-res display and in latency because doing more stuff takes more time.

This is why top racers like Amari fly at 540 resolution for their video. The want to reduce latency, so they fly at 75% less resolution than 1080. The same principle applies. Anything you increase, you need to decrease something else by the same ratio to pay for it.

So you can decide what's actually most important to you.

PR #5090 (2019-10-17): Add support for FrSky's pixel based OSD
Protocol might still receive some additions before release (but not breaking changes).

Configurator PR at https://github.com/iNavFlight/inav-configurator/pull/835

Fixes #4287
Unblocks #3578,  #4210 and #4911