iNavFlight/inav #4295: Betaflight F4 (BFF4) Motor 1 not responding in 2.1 RCs

trekki1978, 2019-01-31:
Hi,
I'm using BFF4 board and flashed 2.1.0-RC1 and also RC2 and found out that Motor#1 does not respond.
After rolling back to version 2.0.1 all Motors work fine.

Marco

shellixyz, 2019-02-02:
Please post a config dump

trekki1978, 2019-02-02:
[INAV_cli_20190202_235955.txt](https://github.com/iNavFlight/inav/files/2824964/INAV_cli_20190202_235955.txt)
I reflashed 2.1.0 RC2 and configured the copter to get this dump file. Still no reaction from motor#1. 
ESC#1 does not show start beeping. ESCs and motors #2, 3 and 4 work fine.

shellixyz, 2019-03-03:
This should fix your issue, please try and report whether the issue is still there: 
[inav_2.1.0_BETAFLIGHTF4.hex.zip](https://github.com/iNavFlight/inav/files/2923615/inav_2.1.0_BETAFLIGHTF4.hex.zip)

trekki1978, 2019-03-08:
Hi,
with this hex file all motors work fine.
But due to another issue with the external Baro I still have to use a customized 2.0.1 firmware on my BFF4.

Thanks so far
Marco

shellixyz, 2019-03-09:
Good to know that at least the motors work now, thank you for the feedback. I see that you already have an issue open for the baro so look there for a follow up on the issue.

PR #4447 (2019-03-09): Fix BETAFLIGHTF4 target motor 1
Fixes #4295

Motor 1 timer definition haven't been migrated correctly from 2.0.1
commit 21bcd50b (2019-03-09): Fix BETAFLIGHTF4 target motor 1 (#4447)

Fixes #4295
closed 2019-03-09 by shellixyz