iNavFlight/inav #4296: OSD locked on NTSC

mariniam, 2019-01-31:
## Current Behavior
Dear all, through my goggles I only see NTSC meaning many labels from the OSD are not displayed as I have set them up in PAL. No matter what i do in iNAV configurator, AUTO or manual PAL/NTSC selection, in my goggles I have NTSC. Even with my Runcam split2s, I go into its config and manually choose PAL, but it switches back to NTSC. Any ideas since I really want to use PAL since you can fit a lot more information on the screen. 

Have tried with 2.1.1 and 2.1.0 configurators
Flashed FC to 2.0.1, 2.1 RC1 and 2.1 RC2 

- FC Board name and vendor: Matek 405 CTR
- INAV version string: 2.0.1 and 2.1 RC1 and 2.1 RC2

Nihilistic, 2019-02-01:
Have you selected PAL in your camera's menu?

What VTX are you using? Is it capable of both NTSC and PAL, and automatically transmits whatever is input to it? Did you first load the font on the OSD tab, and then choose PAL and save? Sometimes the OSD tab comes up blank for me, and I have to close iNAV and restart it and go back to the OSD tab.

I have the Split 2S in PAL mode on my quad with the F405-CTR, and it's working in PAL. I stuck with iNAV 2.1 RC1 rather than upgrade to RC2 because I'm using a custom hex that fixes the Split 2S transmitter control, so I can start and stop recording from an AUX switch. No telling when this fix will be rolled into the releases.

Nihilistic, 2019-02-01:
What goggles and what video receiver? Are you sure it displays PAL?

mariniam, 2019-02-01:
Hi thanks for your input! I'm pretty sure the issue is with my Split 2S and not iNAV after all because I flashed Betaflight on the Matek and Im getting the same behaviour. Everything stuck on NTSC. Problem is the Split 2s has NTSC as default and for some reason is not saving any changes I make in the settings. I choose PAL but it goes back to NTSC as soon as I exit the menu. So im going to try reflashing my Split 2S and hope for the best. Any chance you could share the hex file :-) ?

Nihilistic, 2019-02-01:
Are you sure you're saving the changes after selecting PAL in the Split 2S app? Are you using the wifi app, or the OSD menu? Are you using the latest Split 2S firmware? What version and date?

azolyoung posted the hex in this github issue

[https://github.com/iNavFlight/inav/issues/2747#issuecomment-454344364](url)

[https://s3.amazonaws.com/rctmpfiles/inav_test_fws/inav_2_1_fix_rcsplit_telemetry_control/inav_2.1.0_MATEKF405.hex](url)

"If you still can't enable camera control using the following firmware, please paste the following three commands into the cli interface and try again:"
set rcdevice_protocol_version = 1
set rcdevice_feature = 7
save

I had to enter the CLI commands he suggested after loading his hex, and it works. 

The downside is that we're stuck with RC1, or we'll have to keep bugging him to make another hex for every new RC until it gets rolled in.

mariniam, 2019-02-02:
I flashed the latest runcam firmware and seems to be working now. I can finally save PAL and other settings!  Will give camera control a try via the CLI commands. Thank you so much for helping out. So the issue was not related to iNAV or Matek F405 CTR. case closed

closed 2019-02-02 by mariniam