iNavFlight/inav #4299: Calibrating accelerometer MatekF722

alessandro-neri, 2019-02-01:
Hello, I have a MatekF722 and I can't calibrate the accelerometer. The steps 1 and 2 succeede but the step 3 doesn't work. Once the step 3 fails it seems that the calibration process crashes and I have to retry doing the calibration of the step 2 and 3, not only step 3. I tried with version 2.0.0 and 2.0.1. 
(I am calibrating the accelerometer relative to the board not relative to the model. I have read the sensor calibration documentation.)
How do I solve the problem?
Thank you.

shellixyz, 2019-02-02:
I can't reproduce the issue (2.1-RC2 and latest configurator)

DzikuVx, 2019-02-27:
I'm also unable to reproduce so I suspect it's either user or hardware error

closed 2019-02-27 by DzikuVx