iNavFlight/inav #4300: 2.1 RC2 Decoded log files are corrupted at the end

OlivierC-FR, 2019-02-02:
After being decoded, the last line of the main log contains garbage values, and the last 10% of the line for the GPS are also very wrong.
Below are the 2 CSV exports, and also the source log.
Same result with the iNav decoder, and an old Cleanflight decoder that worked fine with all previous iNav releases.

Blackbox is set as follow :

set blackbox_rate_num = 1
set blackbox_rate_denom = 100
set blackbox_device = SDCARD

Full dump for the FC, a Matek F405-SE : https://www.dropbox.com/s/ka1x6qjkhqgmw9r/DXL.dump.txt?dl=0

[image]

[image]

[LOG90002.zip](https://github.com/iNavFlight/inav/files/2824496/LOG90002.zip)

OlivierC-FR, 2019-02-02:
Decoded files : [logs.zip](https://github.com/iNavFlight/inav/files/2824502/logs.zip)

OlivierC-FR, 2019-02-02:
Good catch, indeed looks like I forgot to disarm. Before I close this report I'll double-check on the bench.
Side note, the blackbox logging on/off control is on the same switch as the arming.

giacomo892, 2019-02-03:
It is not suggested to have other modes on the arming switch beside the ARM one

OlivierC-FR, 2019-02-05:
Closing, plane was unpowered while stil armed

closed 2019-02-05 by OlivierC-FR