iNavFlight/inav #4303: Scrolling side bars

oldrootbeer, 2019-02-02:
Hi.
Is it just me, or do the scrolling side bars scroll the wrong way when ascending / descending? Shouldn't they scroll down with ascension and vice versa?

oldrootbeer, 2019-02-03:
No. I mean the side bars. Not the Vario. I mean these:
[image]

oldrootbeer, 2019-02-03:
Well I thought I'd try anyway. If the side bars had numbers, eg 10m, 20m etc, to indicate height, they would scroll the other way... Close if you wish.
[image]

shellixyz, 2019-02-03:
Indeed I think it would make more sense if the sidebar moved the other way for altitude

teckel12, 2019-02-03:
@shellixyz I'll have to take a look, I assumed it moved the same way a glass cockpit attitude display worked.

oldrootbeer, 2019-02-08:
Only if it's do'able, perhaps have the altitude in 10's of meters on the side like the above pic. Also angle the side bars wrt roll. Not sure if that is easy to code. But would look spot on.

teckel12, 2019-02-08:
@oldrootbeer The problem with changes like you're suggesting is that the OSD is all character based and you can't have characters on top of each other.  So, tilting the side bars would 1) look ugly 2) lose the visibility of a scrolling sidebar 3) probably overwrite other characters.

Trying to do a glass cockpit display with just characters is challenging, and a few sacrifices need to be made.  While your suggestion is sound, it really should be reserved for a future pixel-based OSD (which oddly enough INAV started working on just yesterday).

I'll have to check the scrolling side bars.  But, altitude should scroll down when going up and the speed should scroll down when going faster.  This is the traditional direction for glass cockpit displays.  

Example from INAV Lua Telemetry:

[image]

oldrootbeer, 2019-02-09:
Understood. Should look great if you pull it off.

teckel12, 2019-02-09:
@oldrootbeer I'll look into adding small altitude/speed numbers next to the side bars. But I think it will make it too wide, and lots of pilots put the altilutude/speed/vertical speed to the left or right side of the side bars. But, maybe with the small numbers next to the side bars, the actual altutude/speed won't be as needed? I'll look at the code/screen layout.

shellixyz, 2019-06-09:
@teckel12 Any news about this ? Reversing the direction of the altitude side bar would already be a nice improvement.

teckel12, 2019-06-10:
@shellixyz I'll need to check, I thought the dots crawled down when you gained altitude.

closed 2019-08-23 by stale[bot]