iNavFlight/inav #4304: Feature Upgrade:  Better waypoint tracking/Using tracks instead of solely waypoints

KaisarX, 2019-02-03:
## Current Behavior
Fixed wing craft target a waypoint and fly directly to it.

## Desired Behavior
Pilots attempt to fly TRACKS to waypoints when working with surveys or other kinds of navigation

## Suggested Solution
When determining how best to approach a waypoint, not just the next waypoint needs to be considered but the point after that as well, and the one from where I came.
For example:

 1-----------------------------------------> 2
                                                                   | 
                                                                  \/
4 <----------------------------------------- 3
flying from waypoint 1 -> 2   determine the expected track  in this case it should be bearing 090
but if i'm 10 meters to the right of the line if i'm flying 090, I will fly parrallel and 10 meters off track for the duration of the line.  And in order to perform a good survey I need to get back to the line reasonably quickly.  
Lets say my bearing to the waypoint is 085.  This means i'm far away... but that's not every helpful.  Let's say instead I pick a point along the line 2 seconds ahead of where I am.  Since i'm using GPS and my groundspeed is 15m/s  I choose a point 30meters down the line... what's my bearing to that point.  
In this case it's approx 072
Ok so I'm about 18 degrees away from flying back to the line (90-72)
So correct my current heading to bearing 72.
Do this repeatedly until my desired bearing equals my actual bearing.  ( you could make this return to the line faster by increasing the a return to line factor... ie.. instead of just going
current bearing - bearing to track.  (090-18 = 072)
go -> current bearing - bearing to track * multiplier  (090 - (18 *1.5) = 063)

*NOTE*  higher precision on bearings will allow you to get closer to the line. but in this case 30m down the line at < 1 degree off is a difference of 0.5 meters.

PHASE 2:  
as you approach a waypoint it's time to set a new track
continuing with our previous example:
      speed = 15 m/s
      distance off track = < 1m
      bearing = 090
      distance to next waypoint = < 15m

get position of next two waypoints.  get bearing from waypoint 2 to waypoint 3  
             In this case its bearing 180.
great.  Now, what would be the angle from our current bearing to a spot  2 seconds along that track.
in this case we are approximately 15 meters from the track (heading straight towards it at 090) and we want to go toward a spot 30 meters to the planes right.
Do a little more triangle love like we did before.
15 meters from line
to 30 meters down the track
with a right angle between us
equals a bearing to the target as approximately 63 degrees to the planes right.. which means we add
  *NOTE*:  we add to the right and subtract to the left
so our new desired bearing is 153
(NOW it is important to note that many planes limit their nav bank angle to 45 and thats a good thing.  So instead of bearing 153 we will instead bear 135.  It will still get us to the line... we might blow through the track but in this scenario we probably won't reach the track before we get too close to the next waypoint anyway.)

and since we are continuously trying to recalculate that proper bearing that angle will decrease as we get closer to our desired bearing of 180.

Then rinse and repeat.  When you reach the waypoint threshold just get the next two waypoints calculate a new track and away you go.

## Who does this impact? Who is this for?
This can be used for anyone who wants to fly a more rigid flight pattern.  Convergence can happen within seconds.  By following a track you are essentially holding a carrot(waypoint) on a stick(2 seconds) infront of the horse(plane)
So, grid patterns are cleaner, circle patterns are stuck to like glue, and correction for wind drift is baked right in (if you correct more than a factor of 1... i suggest a factor of 1.5 to test, and to make it adjust for wind see below for notes on determining winds).   Agricultural users, surveyors of all kinds, users of traffic patterns, traveling down UAV corridors, staying away from no fly zones, even geofencing could benefit from this.

## Additional context*
You could potentially use this to determine windspeed and direction while flying grids with this method as well.  
ie. if you are maintaining a bearining and unable to reduce your bearing any further but still maintain a certain distance from the track , then in order to actually return to the track you need to increase your angle towards the line.  
What that looks like is,
plane flying 15m/s, tracking 090, traveling 2m to the right of the line bearing approx 086, and that distance and bearing have become stable, presumably you are facing at least a 2m/2s wind from the left and .067 m/2s from the front 
therefore you have a wind at approx 1 m/s  from 268 and you are actually travelling about 16 m/s

I think my math is right on this part.. but its getting late and I'm getting tired.

Oh I have a formula for finding distance to a line lying around somewhere that doesn't use angles (or at least reduces its use to a minimum) and instead uses certain kinds of adding and multiplication and such if anyone is interested.

OK HAVE FUN! :)

KaisarX, 2019-02-08:
Ok what follows is the steps to determine distance from a track and along a track without trig.  It's a modified work based on complex numbers.  For an excellent read on the subject go over to https://betterexplained.com/articles/understanding-why-complex-multiplication-works/

Step 1.  gather your two waypoints (wp1, wp2) for your desired track and your location. (loc)
step 2.  subtract wp1 from the other two points.   wp2 - wp1 and loc - wp1
step 3.   find the length from 0 to wp2.  And call it magnitude.
step 4.  using complex angles subtract wp2 from loc  
               -------> (locX+locY)*(wpX - wpY)
               -------> locXwpX + locX(-wpY) + (locY)(wpX) - (locY)(-wpY)
               distance along track =>( locXwpX + locywpY  ) / magnitude   (Neg is before waypoint 1)  
               distance from track => (locYwpX - locXwpY) /magnitude     (Neg is to the right of the track)

NOTE:  in the above formulas I did not show all my work, therefore negatives may seem to jump around and it may be confusing as to why you would subtract (locY)(-wpY) and that has to do with imaginary numbers....   when you square an imaginary number you kind of "toggle' between - and positive...  yeah

I digress.
A programmatic way of writing this would look like

`//this will find the distance from the track and along the track with direct decimal latitudes and //longitudes
function find_track(wp1, wp2, location)
{
      wpX = wp2lon - wp1lon
      wpY = wp2lat - wp1lat
      locX = locationLon - wp1lon
      locY = locationLat - wp1lat

      mag = sqrt(wpX*wpX + wpY*wpY)

      distAlongTrack = (locXwpX + locywpY  ) / mag   //(Neg is before waypoint 1)  
      distFromTrack = (locYwpX - locXwpY)  / mag       //(Neg is to the right of the track)
}
`

KaisarX, 2019-02-08:
Now... while I was writing that comment I realized I might as well just do the code for the whole suggestion.  I'm sorry i'm not just putting it straight into a pull and stuff... but I'm a little nervous to break everything else that works so well by not knowing exactly where I should put it.... anyway... here goes... and i haven't done C in a while.  So here's a pseudo code

`function followTrack(wp1, wp2, loc,speed)
{
     wpX = wp2lon - wp1lon
     wpY = wp2lat - wp1lat 
     locX = locationLon - wp1lon
     locY = locationLat - wp1lat
     
     mag = sqrt(wpX*wpX + wpY*wpY)

    distAlongTrack = (locXwpX + locywpY  ) / mag   //(Neg is before waypoint 1)  
    distFromTrack = (locYwpX - locXwpY)  / mag       //(Neg is to the right of the track)

  distFromTrackMeters = distFromTrack*(conversionnumber to convert longitude/latitude to Meters)

  desiredBearing = todegrees(arctan(wpY/wpX))    //if flying from wp1 to wp2 this would be your bearing
     
  //assuming speed in m/s
  turnToBearing = desiredBearing + toDegrees(Arctan((distFromTrack)/speed))
  //alternately you can add in a variable to adjust just how aggressive to attack the track
     turnToBearing = desiredBearing + toDegrees(Arctan(((distFromTrack)/speed)*multiplier))
  //higher multiplier has a more aggresive correction
  
  return turnToBearing
}`

breadoven, 2022-11-26:
Closed by https://github.com/iNavFlight/inav/pull/8234.

closed 2022-11-26 by breadoven