iNavFlight/inav #4309: AnyFC F7 output mapping change on 2.1, No Servo Output 1.9 onwards

ryanort, 2019-02-03:
## Current Behavior
Motor output mapping based on https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20AnyFC%20F7.md changed on FW 2.1 RC1 and RC2.

Servo output based on https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20AnyFC%20F7.md not working starting FW 1.9.  Tested to be still working on FW 1.8

ryanort, 2019-02-10:
tried it again on 2.1.0 RC2 and found out that motor output mapping is ok but not if LED strip support is turned on

ryanort, 2019-03-06:
still no go on 2.1 final release and no one seems to be able to provide any feedback either.

DzikuVx, 2019-03-11:
Servo output will be fixed in #4494
PLease do not use LED strips with this board currently

ryanort, 2019-03-11:
Excellent. Thanks!

2T-GT, 2019-05-31:
I got the same issue, enable the LED strip and motor 4 and 2 turned off. I thought I was being cleaver by adding the LEDs , but it tuned the quad into a lawnmower when I throttled up with only 2 motors :-). 
I was hoping the F7 boards were new enough that lack of support was not going to be a issue when I built this machine 2 years ago. 
would the Matek F722-SE be a supported replacement board?
Got to give the dev team credit, I put the toolchain on my machine and tried move ports, I'm well in over my head with this and my level of understanding C.

golddragon007, 2020-04-05:
@DzikuVx Because it's old hardware it doesn't mean it is a bad too and shouldn't be supported. I selected in the past this controller because it had the possibility to use it with octocopter and LEDs. But please then tell another controller which is supported, it's F7 or better and I can put tons of sensors on it like on this and can be used with octocopter too.