iNavFlight/inav #4317: Can I get Autotrim Values from Log?

b14ckyy, 2019-02-05:
Hi Team, 

Yesterday I made the re-maiden flight of my AR wing with Matek F405 Wing FC. During the flight I used Servo Autotrim but I forgot, that it needed to be enabled during launch (The flight was very unstable due to bad PIFF values and I was flying over a lake so I was a bit nervous). 

I can see the mode change in the logviewer but it is possible to get the detected values from the log too so I can insert them to the FC manually? I only have 4 switches and one potentiometer für extra features and 3 of them are in general use. I'd like to autotune and test some other things on next flight so it would be great if I could avoid just a short trim flight for now.

b14ckyy, 2019-02-05:
Added the Log: 
[LOG00035.zip](https://github.com/iNavFlight/inav/files/2833216/LOG00035.zip)

giacomo892, 2019-02-05:
If you disable autotrim before disarm values will be lost. So you cannot.

DzikuVx, 2019-02-06:
I'm affraid it is not possible

closed 2019-02-06 by DzikuVx