iNavFlight/inav #4324: Brushless Gimbal support through Inav Matek F722SE Lumenier LUX F7

Tazzzzzzzz, 2019-02-07:
I am attempting to run a custom 2x axis gimbal for a small Flir camera on my 12" quadcopter. 

Running Inav 2.1.0 FW on a Lumenier LUX F7 (Matek F722SE) target

Quad is 100%, complete, and functional, sans 2 axis gimbal. attempting to drive the brushless gimbal motors, and cant seem to get the FC to communicate with them correctly.

Lumenier GX2208-80 Brushless Gimbal Motor

**Power/ signal setup to Brushless motors:**
4S Lipo to BL_heli32 ESC's , signal wire from brushless motors connects to ESC, ESC signal connects to PWM output (S7 on FC)  *** Using D-shot600 protocol for Quad motors, I believe this is separate from how the Servo Mixer operates, but unsure...

INAV mixer is set to pads S7, S8 to control Servos 0 and 1 (gimbal roll, and gimbal pitch assigned)

Mode is enabled to run in all positions, Should be active 100% of the time. 

power up brushless motors and FC (Brushless motor audible fire up, no arm sound)
Proceed to motor tab, "I understand risks etc.." power motors..... nothing, no actuation or zeroing of motor. No resistance, or power other than initial startup from lipo.

Tilt quad on the bench roll and pitch, sensor data shows servo's 1 and 2 ( 0 and 1 in mapping) shows articulation, gives corresponding value, no power driving the brushless gimbal motor, no response.

Motor works, when I plug 3S directly to motor, full engagement and position hold, Just cringing on my part for maxing it out. 

Configurator shows correct mapping to S7 and S8 pads for Servo mapping, however when i do a resource command in CLI, it shows motors 7, and motor 8 (Not Servo as described in Configurator)

Attempt to do a resource command to assign Servo 0 to S7, and Servo 1 to S8, wont do it. Somehow CLI remapping in general is not working on this setup.   

I'm to the point, where I would imagine that the FC is attempting to communicate with pads S7 and S8 as it would a regular servo motor, and not a brushless motor through an ESC.

I don't know enough about protocols to estimate what might be going on there, but i'm to the point of ordering another 30.5x30.5 Gimbal controller board to my stack. Seems like a waste with an F7 FC that has the PWM outputs and the software appears to be capable of doing this. 

There is some small disconnect in the firmware. PWM outputs are available (Pads S5,S6,S7,S8) it seems to be setup correctly, but no communication, or incorrect communication is ocurring.

Thoughts, or suggestions?
-Thanks

[INAV_cli_12in 6Feb19.txt](https://github.com/iNavFlight/inav/files/2838753/INAV_cli_12in.6Feb19.txt)

wx4cb, 2019-02-07:
a couple things i see right off.... 

there is no resource remapping in inav

try using pwm instead of dshot for your gimbal motors and make sure you calibrate them.  if that works, then try going up in protocol to multishot. from your description, it sounds like it's trying to send the data to the esc etc, but it sounds like the gimbal isn't recognising the dshot signal.

Tazzzzzzzz, 2019-02-07:
Attempted PWM, and Multishot (With Calibration)

No joy, All quad motors gtg, but not even sufficient power supplied to Brushless gimbal motor.

I do not ever get an audible arm sound from the Brushless gimbal motor. Just power up beep.

Looking at the configurator, the Servo portion is quite separate from the motor portion. 

I can assign Pads for motors 5 and 6 and attempt to connect the esc signal wire there, however on the assignment of motors, I have no function settings (Input dropdown menu) from Servo side. 

I'd imagine the communication line would be correct then (pads S5 or S6), without the ability to add any function to it....

For the servo functions I am limited to pads S7 and S8. which apparently don't communicate in the correct protocol?

I'm thinking that I could run a standardized box style servo from here (Fixed wing flaps), but not brushless motor control?

Seems like a weird gap in the firmware integration to not be able to communicate that function dropdown via the brushless motor controls for surplus PWM outputs (S5, S6)

wx4cb, 2019-02-07:
usually - at least on the 2 brusless gimbals i have, there's a separate motor controller that takes the pwm in from the FC and does all that you need to work the gimbal.

as far as a regular servo gimbal, they work just as regular servos and you add them in on the mixer as gimbal roll etc..

Tazzzzzzzz, 2019-02-07:
ya, that's the next stage in this. Controller should be in the next day or two. 

I just wanted to run to ground the option of using the FC's available PWM outputs since it has the processing power.

Seemed feasible since all the processes etc are so close between driving the motors/ servo capabilities. 

Maybe down the road for Inav and FC's...

DzikuVx, 2019-02-07:
I'm afraid it does not work like that. INAV is not a brushless gimbal controller and can not directly drive any motors. Output of gimbal pitch and roll is actual UAV attitude, not signal to ESC.
It works with servos because servos "know" their output shaft position. Motors do not.
For what you want to do you need external brushless gimbal controller with its own gyro. Signal from FC can be used to help gimbal controller, but not replace it

closed 2019-02-07 by DzikuVx

Tazzzzzzzz, 2019-02-07:
Sounds like plan B, Gimbal controller it is. 

Thanks for the help everyone.

holoduke, 2019-08-21:
Wondering how the setup is when using a alexmos gimbal controller. Can you just forward servo X to input of the gimbal controller ?