iNavFlight/inav #4325: hexacopter motor 5

DALESRC, 2019-02-07:
I using your Matek F722-SE on a hexacopter running Inav unstable firmware (2.1.1 RC2) and I cant get motor 5 to work. So I loaded betafight and it and it worked great. 

# diff

# version
# INAV/MATEKF722SE 2.1.0 Jan 22 2019 / 09:42:23 (7bdd5967e)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer
mmix 0  1.000 -0.500  0.866  1.000
mmix 1  1.000 -0.500 -0.867  1.000
mmix 2  1.000  0.500  0.866 -1.000
mmix 3  1.000  0.500 -0.867 -1.000
mmix 4  1.000 -1.000  0.000 -1.000
mmix 5  1.000  1.000  0.000  1.000

# servo mix

# servo

# feature
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 5 32 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1125

# adjrange

# rxrange

# osd_layout

# master
set acc_hardware = MPU6000
set acczero_x = 107
set acczero_y = 28
set acczero_z = -170
set accgain_x = 4082
set accgain_y = 4100
set accgain_z = 4007
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = ADC
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set model_preview_type = 10

# profile
profile 1

# battery_profile
battery_profile 1

giacomo892, 2019-02-07:
Have you tried not using DSHOT?

DALESRC, 2019-02-07:
no but when i put motor 5 on the broad pad 6 it will work and I have betafight on DSHOT and it all works
 thanks

giacomo892, 2019-02-07:
Try without DSHOT on INAV, thanks.

DALESRC, 2019-02-07:
change DSHOT to what????

and I don't know what pitot is

thanks

giacomo892, 2019-02-07:
to ONESHOT125 for instance.. .

DALESRC, 2019-02-07:
OK i will try that

DALESRC, 2019-02-07:
will that fixed it I don't understand 

thanks for you help

closed 2019-04-22 by stale[bot]