iNavFlight/inav #4329: Inav with Omnibus F4V5 need more UARTs help to remap

Tillidan, 2019-02-08:
Hello,

I'm using this board (omnibus F4V5) and I wired a GPS M8N with magnetometer (uart 1 and 6), a PPM receiver (uart1), and a Bluetooth module (uart 3), to connect to Mission Planner. Also I use a LED strip.

I saw that I need an aditional port, because now GPS a PPM receiver is in the same port... I read something about remapping, Maybe is possible to remap some Pins or resources to use all my hardware? I need some help in that. It is possible to reuse a SERIAL_RX pin to SERIAL_TX pin?
 In that case I could use UART1 rx pin to receive GPS data, and UART1 TX pin converted to "RX" to receive PPM data.

Thanks in advance. to use INAV we need more ports!
Regards,
Javier.

wx4cb, 2019-02-08:
the compass usually goes on the SDA/SCL lines not a uart

Tillidan, 2019-02-08:
Hello, scl and sda shares pins with uart3 and I wired it here. Can I use uart3 with other pins? Remapping them? People using inav I guess they are usually using GPS, compass, receiver and a Bluetooth module to connect, some solution I need...

teckel12, 2019-02-08:
There's no remapping in INAV. Use soft serial if available or try using an F7 FC with more ports. The only other option is to build your own custom FC firmware.

Tillidan, 2019-02-08:
Hello, How can I use soft serial in that case? Thanks. If there is some example I would be appreciated.

teckel12, 2019-02-08:
@Tillidan Depends on if your FC has a soft serial port. Some Omnibus FC have pins exposed, some don't. See your FC documentation or look in the configurator if soft serial is listed with your other UARTs.

closed 2019-02-14 by DzikuVx