iNavFlight/inav #4330: Inav Autotune related Crashes

MadMiky, 2019-02-08:
I tried Autotune two times.
The first time I misunderstood the wiki and did rolls and loops while in Autotune which lead to a bad crash. ( Maybe someone should clarify the autotune section, because in Acro hard manovers mean also loops and rolls to most people.

The second time I got it to work correct, landed and saved the tune.
[image]
But on the next start my plane climbed normal in NAVLaunch but as soon as it changed to RTH it started oscillating and crashed straight into the ground.
Sadly I only have the Black-box recording but no video.

[LOG00010.TXT](https://github.com/iNavFlight/inav/files/2845839/LOG00010.TXT)

It doesnt look like a stall or something, rather it was overreacting and then decided aiming for the ground is a good idea.
RTH was tested and used in the flights bevore.

I used the last Inav version available 2.1.1 ore something
Is it save to use this feature?
I´m afraid that my AR-Wing cant handle more crashes like this.

IVData, 2019-02-13:
> 2.1.1 ore something

To receive some help, you really need to be sure about the version. 2.1 is not a stable release, but 2.0.1 is.

teckel12, 2019-02-13:
@MadMiky There more details that are important, those pesky things asked when you opened the ticket. For example, supplying the CLI dump will show not only your settings but also your target and the version of INAV your running.

closed 2019-03-01 by shellixyz