iNavFlight/inav #4337: Wing Fc-10 DOF  ppm not working

jazavac2303, 2019-02-10:
The board run's inav_1.9.0_FF_F35_LIGHTNING.HEX 
and the ppm doesn't work. 
I tryed new firmware but no changes.
SBUS work but I need ppm for my long range receiver.
Please help.

giacomo892, 2019-02-10:
Try 2.1 and if you dare sending one of those boards to a developer he can take a look.

jazavac2303, 2019-02-10:
I tried 2.1 and all the versions from 1.9 to 2.1 and also the beta 2.1 R1 and R2.
PPM does not work, only SBus.
I tried arduplane fimware from arducpter for this borad which is the same as Furious FPV F-35 Lightning 
In this arduplane firmware the ppm works normally but I don't really like the layout and I'm used to INAV  and like how it works.
I have only one board and it is mounted on my rc wing so for now I don't plan to send it to a developer.
I will probably run it on Sbus for now and then we will see.
It is definitly some with inav firmware for this board since on arducopter firmware the PPM works.
Thanks for response.

jazavac2303, 2019-02-10:
here is the link for the board:

https://www.banggood.com/Wing-FC-10-DOF-Flight-Controller-INAV-OSD-Accelerometer-Barometer-Gyro-Compass-For-RC-Airplane-Drone-p-1318626.html?rmmds=search&cur_warehouse=CN

shellixyz, 2019-02-10:
Try to wire PPM on UART3 TX it should work

jazavac2303, 2019-02-10:
You mean to rx3?
Ppm is connected to uart3 rx3.

shellixyz, 2019-02-10:
No I mean TX. FF_F35 is configured to use PB10 as input which is UART3 TX

jazavac2303, 2019-02-10:
But on this board wing fc 10 there is no uart3 tx pin.
I can't find it.
Or maybe is different labelled?

jazavac2303, 2019-02-10:
I will take a look at STM32F405RGxx pin out and see where the pb10 is connected. 
Thanks for information.

shellixyz, 2019-02-11:
Indeed there is not UART3 TX. Try with this build which puts the PPM input on UART3 RX:  [inav_2.1.0_WINGFC.hex.zip](https://github.com/iNavFlight/inav/files/2849609/inav_2.1.0_WINGFC.hex.zip)

jazavac2303, 2019-02-11:
This is it, now ppm works.
Thank you so much for this custom firmware.
Big thumps up for shellixyz

shellixyz, 2019-02-11:
Nice :)

I'm reopening because we are going to make a new target suited for this FC

jazavac2303, 2019-02-11:
That is great.
If I find some other issues related to this board I will post here so it can be fixed.
It is really hard to keep track of so many boards and clones of boards.
Thanks again for the very quick respond and fix.

digitalentity, 2019-02-18:
Closing, we now have a separate target for this thing

PR #4348 (2019-02-12): Add new target WingFC
Where to buy:
https://inavflight.com/shop/s/bg/1318626

Closes #4337
commit b9930dec (2019-02-12): Add new target WingFC

Where to buy:
https://inavflight.com/shop/s/bg/1318626

Closes #4337
closed 2019-02-18 by digitalentity