iNavFlight/inav #4340: autoconnect doesn't work on F405-ctr

laurentopia, 2019-02-10:
clicking on the connect button works though

teckel12, 2019-02-11:
@laurentopia That would be a configurator issue, not INAV. Please open an issue in that repo. But for what it's worth, I've found this feature hit or miss on many flight controllers and in Betaflight and INAV. And really, it's just as easy to click connect when you want to.

laurentopia, 2019-02-12:
Yes it is minor. I didn't know it was common to FC, is this a hardware side glitch?

teckel12, 2019-02-12:
@laurentopia Not sure, probably a glitchy timing thing. Like some FC, ports or drivers take a bit longer than expected. Still, this issue should be on the configurator repo, not here.

DzikuVx, 2019-02-27:
Please do not use Autoconnect on configurator. It will not correcly work with F4/F7 boards using VCP USB connection. The fact that port appeared in the list does not mean board booted up and is ready so serve requests. Autoconnect will be probably removed since reconnect was fixed and configurator correctly reconnects to FC after reboot

closed 2019-02-27 by DzikuVx

laurentopia, 2019-03-01:
Great to hear you're removing busted features.