iNavFlight/inav #4341: 2.1 RC3 when using IMPACT font  OSD is either blank or shows none of the options

laurentopia, 2019-02-10:
NOTE: all the options selected previously show in the goggle
the left portion of the OSD GUI some times doesn't show at all after reboot or the options previously selected are left blank, the preview is blank too, only the background image is shown.

laurentopia, 2019-02-10:
Also happens after restoring to default font.
On a matek f405-ctr, erased an flashed

kardon18, 2019-02-11:
I found the same thing.
The way I got around it for now.
Was to load the font you require. Let the FC reboot.
Then power down the FC, disconnect it from the PC, and close the configurator.

Then re-open the configurator, then power up then FC again and re-connect it to the PC.

If the FC doesn't need external power and can be powered from the USB port. Then you don't have to power it externally in above steps. Just connecting it to USB will do the same.

After this. The loaded OSD font and pages can be viewed.

kardon18, 2019-02-11:
I found the same thing.
The way I got around it for now.

1) Was to load the font you require. Let the FC reboot.
2) Then power down the FC, disconnect it from the PC, and close the configurator.
3) Then re-open the configurator, then power up the FC again and re-connect it to the PC.

If the FC doesn't need external power and can be powered from the USB port. Then you don't have to power it externally in above steps. Just connecting it to USB will do the same.

After this. The loaded OSD font and pages can be viewed in the configurator.

You might want to change the title. Because it affects all fonts.

teckel12, 2019-02-11:
@laurentopia Also, you may want to go to the configurator repo and look in the issues there. This repo and issues are for INAV, not the configurator.

laurentopia, 2019-02-12:
Thanks for the working steps
> You might want to change the title. Because it affects all fonts.
done

> @laurentopia Also, you may want to go to the configurator repo and look in the issues there. This repo and issues are for INAV, not the configurator.

Is it possible for the OP to move a thread to another repo?

kardon18, 2019-02-12:
Looks like @DzikuVx has seen the issue you posted here. And has a PR for it.
https://github.com/iNavFlight/inav-configurator/pull/688

closed 2019-02-27 by DzikuVx