iNavFlight/inav #4343: Suggesting "PREPARING TO LAND" only at zero speed

sincoder, 2019-02-11:
## Current Behavior
 "PREPARING TO LAND" above home point, But do not consider the speed 
## Desired Behavior
quadcopter stop at home point and "PREPARING TO LAND""

## Suggested Solution
fly near home point , ALTHOLD + POSHOLD and wait it speed to zero or below a certain threshold and distance not far from home point and start  "PREPARING TO LAND""

## Who does this impact? Who is this for?
quadcopter

## Additional context
speed too fast to landing and I have to cancel landing to prevent it from landing elsewhere
[image]

teckel12, 2019-02-11:
@sincoder To me, the question is why is it going so fast when it's preparing to land? My quad flys overhead, becomes stationary, then descends to land. There's basically no forward speed. Is it toilet-bowling? What RTH method have you selected? Maybe your CLI dump?

sincoder, 2019-02-11:
# dump

# version
# INAV/OMNIBUSF4V3 2.1.0 Feb  4 2019 / 15:57:56 (ede4a7904)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature THR_VBAT_COMP
feature VBAT
feature TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature GPS
feature CURRENT_METER
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
[... 877 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 15
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 460
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

sincoder, 2019-02-11:
maybe my RTH speed  or landing speed too fast, But I think it still need a mechanism of protection

kardon18, 2019-02-11:
I agree with @teckel12 
The only time I have experienced your situation @sincoder. Is if NAV_RTH is trying to land the quad in a very strong cross wind.
Under this condition the quad will keep getting blown off course, from the home descent position.
 
I noticed that you have made changes to the Barometric Attitude Hold PID setting from default. 
The changes you made on a few are rather drastic. This could be adding to your problems.

            
```
set nav_mc_pos_z_p = 60
set nav_mc_pos_z_i = 0
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 80
set nav_mc_vel_z_i = 35
set nav_mc_vel_z_d = 14
set nav_mc_pos_xy_p = 60
set nav_mc_pos_xy_i = 45
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 35
set nav_mc_vel_xy_i = 30
```

That combine with your - `set nav_auto_speed = 2000` 
Which is around 72Km/h.

Also, PosHold and Altitude Hold are not required it be selected under the modes tab, in conjunction with RTH.
In fact I found it caused issues.

kardon18, 2019-02-11:
I agree with @teckel12 
The only time I have experienced your situation @sincoder. Is if NAV_RTH is trying to land the quad in a very strong cross wind.
Under this condition the quad will keep getting blown off course, from the home descent position.
 
I noticed that you have made changes to the Barometric Attitude Hold PID setting from default. 
This could be adding to your problems.

            

That combine with your - `set nav_auto_speed = 2000` 
Which is around 72Km/h.

Also, PosHold and Altitude Hold are not required to be selected under the modes tab, in conjunction with RTH.
In fact I found it caused issues.

sincoder, 2019-02-11:
@kardon18 Thank you for your analysis. The wind of that day is about 4-5M/s. I prefer to turn off automatic landing in future flights. RTH with POSHOLD + ALT HOLD  seems to be no problem， I always do it like this

kardon18, 2019-02-11:
> 
> 
> @sincoder  RTH with POSHOLD + ALT HOLD seems to be no problem， I always do it like this

The reason I mention it. Was because of a BAD issue that I had, only a couple of months back. Maybe one time in five, when NAV RTH was triggered, with Alt Hold also selected.

When the quad was moving forward and Failsafe Nav RTH was engaged; it would slow down and turn towards home, to fly back. But it was also trying to hold altitude at the same time. This caused it to violently flip over and lose altitude, before it would self right and stabilize again.
If it had been too close to the ground it would have crashed before it recovered.

It may not affect you...But just be cautious, Because if it does this. Then you'll know what caused it ; )

teckel12, 2019-02-11:
@sincoder While it shouldn't matter, there's no reason why POSHOLD & ALT HOLD should be on during RTH. I can see this potentially creating an edge case situation that goes untested during development.

Anyway, your return speed is probably too high. I think mine is set to a more realistic 500.

Also, with extra arming safety off, you're risking not knowing where home is to return to. It could easily try to return to null island at 72kph with your current setup.

sincoder, 2019-02-11:
@teckel12 Thank you for your suggestion I will test these on my new quad. The 20m/s not the real RTH speed , the max angle 25 will limit it to about 15m/s ， very efficient speed .

digitalentity, 2019-02-11:
With `set nav_auto_speed = 2000` you instruct the firmware to return to home at 20m/s. When you also limit the angle to 25deg (limiting the quad speed to 15 m/s) you are putting firmware in the state when it is unable to reach the required return speed. This may cause unexpected behavior.

sincoder, 2019-02-13:
@digitalentity  Even if I set the speed to 5m/s, it is still possible can not reach the required RTH speed due to big wind . or Maybe it is a compass issue, combine this issue with https://github.com/iNavFlight/inav/issues/2927

sincoder, 2019-02-13:
if INAV have HOLD status ( or sensor status) detection mechanism and auto exit  to lower flight mode 
 would be enough ,  refer #2927

closed 2019-02-13 by sincoder