iNavFlight/inav #4344: Impossible to activate DSHOT

BariMc, 2019-02-11:
Hello,
I have a Kakute F7 and can not activate the DSHOT with version Inav 2.1 RC3, the same for the RC1 and RC2. When I try to activate it and I save and restart, the DSHOT is not taken into account and it restores the default esc settings. Do you have a solution ?
Thank you

DzikuVx, 2019-02-11:
Release notes states that:

DSHOT is enabled on following targets:

AIRBOTF4
ASGARD32F4
ASGARD32F7
COLIBRI_RACE
FIREWORKSV2
FOXEERF405
FOXEERF722DUAL
MATEKF405 (CTR, STD, not WING)
MATEKF722
MATEKF722SE
OMNIBUSF4
OMNIBUSF7NXT
REVO

closed 2019-02-11 by DzikuVx

BariMc, 2019-02-11:
Thank you for this quick response, is this definitive or will be good in version 2.1 stable?

THX

DzikuVx, 2019-02-11:
Probably none of te devs have tis board to test if it does not break anything